aritchie/bluetoothle #168: CrossBleAdapter.Current.WhenStatusChanged() is called twice.

kikutano, 2018-04-16:
## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [] iOS
 - [x] Android
 - [] tvOS
 - [] macOS
 - [] .NET Standard
 - UWP IS NOT SUPPORTED DURING THE BETA PERIOD - ONLY PRs ARE WELCOME FOR NOW

## Version of OS(s) listed above with issue

Android 8.0.1

## Version of Library
5.3.5

## Expected Behaviour
The code internal at CrossBleAdapter.Current.WhenStatusChanged().Subscribe() is called one time when I PowerOff Bluetooth and one time when i PowerOn Bluetooth on my Device.

## Actual Behavior
The code internal at CrossBleAdapter.Current.WhenStatusChanged().Subscribe is called twice, two time for PowerOff, two time for PowerOn.

I call **_just one time_** this code:

`CrossBleAdapter.Current.WhenStatusChanged().Subscribe( statusResult => {
                Console.WriteLine( ">[BLEManager]: BLE Status Changed in: " + CrossBleAdapter.Current.Status.ToString() );

				switch ( statusResult ) {
                    case AdapterStatus.Unknown:
                    case AdapterStatus.Unsupported:

                    Console.WriteLine( ">[BLEManager]: Adapter Status Unkwon or Unsupported." );

                    return;

                    case AdapterStatus.PoweredOn:

                    m_BluetoothPowerOn = true;

                    break;

                    case AdapterStatus.PoweredOff:

                    m_BluetoothPowerOn = false;

                    break;
                    case AdapterStatus.Unauthorized:

                    break;
                }

				OnBLEStatusChangedCallback( statusResult );
            } );
        }`

For some reason, this Subscribe is fired up two time when i PowerOff the Bluetooth and two time when I PowerOn my bluetooth. Is this a bug? On iOS it's called just one time, but on Android it's called twice. I don't know why!

Thanks!

kikutano, 2018-04-16:
I tried on Android 6.0.1 and when I PowerOff the Bluetooth a crash occur:

`[BluetoothGatt] android.os.DeadObjectException
[BluetoothGatt] 	at android.os.BinderProxy.transactNative(Native Method)
[BluetoothGatt] 	at android.os.BinderProxy.transact(Binder.java:503)
[BluetoothGatt] 	at android.bluetooth.IBluetoothGatt$Stub$Proxy.clientConnect(IBluetoothGatt.java:909)
[BluetoothGatt] 	at android.bluetooth.BluetoothGatt.connect(BluetoothGatt.java:771)
[BluetoothGatt] 	at mono.java.lang.RunnableImplementor.n_run(Native Method)
[BluetoothGatt] 	at mono.java.lang.RunnableImplementor.run(RunnableImplementor.java:30)
[BluetoothGatt] 	at android.os.Handler.handleCallback(Handler.java:739)
[BluetoothGatt] 	at android.os.Handler.dispatchMessage(Handler.java:95)
[BluetoothGatt] 	at android.os.Looper.loop(Looper.java:148)
[BluetoothGatt] 	at android.app.ActivityThread.main(ActivityThread.java:5451)
[BluetoothGatt] 	at java.lang.reflect.Method.invoke(Native Method)
[BluetoothGatt] 	at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:726)
[BluetoothGatt] 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:616)`

aritchie, 2018-04-16:
I can't reproduce either.  Do the samples do this for you as well?

kikutano, 2018-04-16:
I can't compile the sample right now, but the I build a simple raw project to test this issue, the Code is here: https://pastebin.com/4YA40YQN and the problem it's the same. Another problems is, when the device reconnect the Notifications are not fired. 

`charaResult.EnableNotifications( true ).Subscribe( result => {
 
                        Console.WriteLine( "\n***************************************************************\n" );
                        Console.WriteLine( "\n*****>[BLEManager]: EnableNotifications OK.********************\n" );
                        Console.WriteLine( "\n***************************************************************\n" );
 
                        charaResult.WhenNotificationReceived().Subscribe( notification => {
 
                            Console.WriteLine( "Misurazione!" );
                        } );
 
                    } );`

But If the device disconnect and reconnect again, for the third time, the notifications start so fire again! I really don't know where I'm wrong.

aritchie, 2018-04-16:
I don't appreciate being bombarded with several different issues in 1 thread along with a wad of code.  Please note that this isn't tech support - sometimes you'll need to get your hands dirty as well if you want help.

You started with adapter on/off - is it a downstream effect for the additional issues?  Where is the bluetooth crash coming from?  I can't see any of my code path to help and I can't reproduce any of the issues you've brought up