aritchie/bluetoothle #170: No Platform plugin found after update to 6.0.0-beta5

mstein11, 2018-04-17:
## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [] iOS
 - [X] Android
 - [] tvOS
 - [] macOS
 - [] .NET Standard
 - UWP IS NOT SUPPORTED DURING THE BETA PERIOD - ONLY PRs ARE WELCOME FOR NOW

## Version of OS(s) listed above with issue
6.00

## Version of Library
6.0.0-beta5

## Expected Behaviour
After adding the nugget package to the android project, it should not throw the error: No Platform Plugin found. Did you install the nugget package into the app project as well.

## Actual Behavior
It does throws said error. some more information:

- Linker is disabled and I cleaned and manually deleted the bin/obj folder before trying
- I update from 5.3.5 to 6.0.0-beta5, afterwards the error occurs
- I am using a SharedProject but the error occurs also if I access CrossBleAdapter.Current from within the android project

## Code of Conduct
* No Library Update Requests
* No Timeline Requests
* Please read through the documentation and FAQs
* Please be respectful of my time. I put a lot of effort into my libraries.  I try to support the issue requests within a day.  I'm not psychic, so please take the time to fill out the above form as complete as possible

aritchie, 2018-04-17:
With betas, I ask users to do a bit of their own digging and contribute back.  I’m already working on the beta and doing testing.  I need help, not support requests.  Sorry

closed 2018-04-17 by aritchie

mstein11, 2018-05-18:
Hi aritchie,

the issue still occurs with the latest stable release (6.0.3). I found the cause of the issue and a workaround:

After updating from a 5.x.x Version of the library to Version 6.0.3, the csproj file of the android project has the following reference: 
```
<Reference Include="Plugin.BluetoothLE">
   <HintPath>..\packages\Plugin.BluetoothLE.6.0.3\lib\netstandard2.0\Plugin.BluetoothLE.dll</HintPath>
</Reference>
``` 

That is, why `CrossBleAdapter.Current` is not properly initialized. By manually changing the line to the following the issue seems to be solved:

```
<Reference Include="Plugin.BluetoothLE">
   <HintPath>..\packages\Plugin.BluetoothLE.6.0.3\lib\monoandroid80\Plugin.BluetoothLE.dll</HintPath>
</Reference>
```

aritchie, 2018-05-18:
You would only see those references if your project isn’t using <PackageReference>.  You need to convert from packages.config and that will solve your issue