aritchie/bluetoothle #171: Duplicated characteristics

ThiagoAguirre, 2018-04-19:
## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [] iOS
 - [x] Android
 - [] tvOS
 - [] macOS
 - [] .NET Standard
 - UWP IS NOT SUPPORTED DURING THE BETA PERIOD - ONLY PRs ARE WELCOME FOR NOW

## Version of OS(s) listed above with issue
5.0

## Version of Library
5.4.5

## Expected Behaviour
1 - Device discovery
2 - Connect
3 - Characteristic discovery
4 - Subscribe to discovered characteristics
--/ The remote bluetooth device is intentionally powered off /--
5 - Start over 

## Actual Behavior
The first connection works flawlessly.
After the BLE device is powered off and on again, on step 3 the characteristics are duplicated.
The disconnect handler calls .dispose() on all previous used functions.
If I power cycle the device again, everything works flawlessly, then again and I get duplicated characteristics, the pattern repeats, works, doesn't work, work...

## Code of Conduct
* No Library Update Requests
* No Timeline Requests
* Please read through the documentation and FAQs
* Please be respectful of my time. I put a lot of effort into my libraries.  I try to support the issue requests within a day.  I'm not psychic, so please take the time to fill out the above form as complete as possible

aritchie, 2018-04-22:
I believe issues around this have been corrected in v6.  Please make sure to submit full samples in the future.

closed 2018-04-22 by aritchie

aritchie, 2018-04-23:
Ok this is by design.  Please read the section under "GENERAL RULES TO FOLLOW".  The very first line states: "DO NOT reuse services, characteristics, and descriptors between connnections"

I supply "Device.WhenAnyCharacteristicDiscovered()" which will fire every (re)connection.

ThiagoAguirre, 2018-04-23:
Yes, I've read the "GENERAL RULES TO FOLLOW"
On each reconnect I'm creating a new device, a new service and reading again the characteristics from this new service.
Nothing is reused. All properties are recreated.

aritchie, 2018-04-23:
Do you observe this same behavior with the sample application?

ThiagoAguirre, 2018-04-24:
Allan, I've reviewed the sample code and rewrote my code do to work the same way as your sample. Now everything is working. Thank you so much for you time. Best regards.

aritchie, 2018-04-24:
Glad to hear it is working.