aritchie/bluetoothle #172: Adapter status is Unknown

bluekuen, 2018-04-23:
## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [x] iOS
 - [] Android
 - [] tvOS
 - [] macOS
 - [] .NET Standard
 - UWP IS NOT SUPPORTED DURING THE BETA PERIOD - ONLY PRs ARE WELCOME FOR NOW

## Version of OS(s) listed above with issue
11.3

## Version of Library
6.0

## Expected Behaviour
Scanning

## Actual Behavior
'''System.ArgumentException: Your adapter status is Unknown
throwing Line:
            bleAdapter.Scan().Subscribe(scanResult => 
            {
                Debug.WriteLine("ScanResult changed: " + scanResult.Device.Name);
            });

bleAdapter is:
IAdapter bleAdapter = CrossBleAdapter.Current;

## Code of Conduct
* No Library Update Requests
* No Timeline Requests
* Please read through the documentation and FAQs
* Please be respectful of my time. I put a lot of effort into my libraries.  I try to support the issue requests within a day.  I'm not psychic, so please take the time to fill out the above form as complete as possible

aritchie, 2018-04-23:
You are scanning before the adapter is ready to take requests.  This is why I provide Adapter.WhenStatusChanged.  ScanExtra works on top of this which is why it works.  Are you sure the 2nd scan exception is "Your adapter status is Unknown".  I'd expect it to possibly be "You already have a scan in progress" or something similar.  I notice you also aren't storing the subscription to your Scan call either.  If you are manually cancelling scans, you need to dispose of the subscription before starting a new one.

Please try to give code external to my library.  Simply reiterating stuff that I've seen isn't helpful.

bluekuen, 2018-04-23:
Hallo Ritchi, Thanks for your fast Response! 
You are right, the 2nd scan exception is: "There is already an existing scan".
I don't store the scan subscription because i'm using takeUntil to automatically get rid of the subscription.

Let me try to explain what i'm doing:
I have a ViewModel called DeviceListViewModel with a simple ObservableCollection of Custom Ble Devices and a Scanbutton in the toolbar. Now, when pressing the ScanButton, i clear the Device List, make sure that Bluetooth on the Mobile Device is enabled and then i call the Scan Method from my
Bluetooth Manager class, which is a simple wrapper for bluetooth related functions (For Scan i just filter for my custom devices). 

When pressing the Scan Button in the toolbar, below method gets called:

```
async Task ScanButton_ClickedAsync()
        {
            if (_scanSub != null)
            {
                _scanSub.Dispose();
            }
            LabelNoDevicesIsVisible = false;

            Device.BeginInvokeOnMainThread(() => UserDialogs.Instance.ShowLoading());

            DeviceList.Clear();            

            SelectedDevice = null;

            var mobileDeviceManager = DependencyService.Get<IMobileDeviceManager>();

            if (!mobileDeviceManager.IsBluetoothEnabled())
            {
                var confirmationResult = await UserDialogs.Instance.ConfirmAsync("Enable Bluetooth?", "Bluetooth disabled", "Ok", "Cancel");
                if (confirmationResult)
                {
                    var deviceManager = DependencyService.Get<IMobileDeviceManager>();
                    if (deviceManager.EnableBluetooth())
                    {
                        UserDialogs.Instance.Toast("Bluetooth has been enabled");
                    }
                    else
                    {
                        Device.BeginInvokeOnMainThread(() => UserDialogs.Instance.HideLoading());
                        LabelNoDevicesIsVisible = true;
                        return;
                    }
                }
                else
                {
                    Device.BeginInvokeOnMainThread(() => UserDialogs.Instance.HideLoading());
                    LabelNoDevicesIsVisible = true;
                    return;
                }
            }
            Debug.WriteLine("Bluetooth Enabled ");

            _scanSub = BluetoothLEManager.Instance.Scan(TimeSpan.FromSeconds(3))
                .Select(scanResult => new CustomBleDevice(scanResult.Device, scanResult.AdvertisementData))
                .Finally(() => {
                if (DeviceList.Count == 0)
                {
                    LabelNoDevicesIsVisible = true;
                }
                Device.BeginInvokeOnMainThread(() => UserDialogs.Instance.HideLoading());
            }).Subscribe(customBleDevice =>
            {
                if (PersistenceManager.GetMyDevicesList().Where(device=>device.ID == customBleDevice.ID).Count() != 0)
                {
                    zaBleDevice.ListStatus = "added";
                }
                DeviceList.Add(customBleDevice);
            });
        }
```

The Scan Method in the Bluetooth Manager class just filters the devices:
```
            var serviceUuidList = new List<Guid> { SERVICE_ID };

            return CrossBleAdapter.Current.ScanExtra(new ScanConfig { ServiceUuids = serviceUuidList })
                .Distinct(sr => sr.Device.Uuid)
                .TakeUntil(Observable.Timer(timeSpan))
                .Finally(()=> { logger.Debug("Stop Scan");});
```

I tried with and without stop scan, but on the second Scan i get, like you said, "There is already an existing" scan.
I hope it is more clear now. Looking forward for your response.

aritchie, 2018-04-23:
The auto-dispose methods don't always fire right away.  You really should manage those disposals yourself.  Also, iOS may not throw the breaks on scanning immediately, so you may need to give a brief pause before starting your next scan.

PS - you don't need to do this.  I manage the thread for you in my other library (stay on topic for this library though pls):
Device.BeginInvokeOnMainThread(() => UserDialogs.Instance.ShowLoading());

bluekuen, 2018-04-23:
I completely refactored my Code so that every subscription gets disposed but it's still not working even after waiting a couple of minutes before scanning again.

aritchie, 2018-04-23:
I guess you'll have to debug with the library.  I run all of my unit tests from a fresh (scan, find device, stop scan, connect to device, do test) and they all run fine.  

Please don't just add "it's not working" as it adds no value to the discussion.  What did you refactor, how, did you look at the lib code to see what may be tripping up for you, etc.

bluekuen, 2018-04-24:
You are right, "it's not working" doesn't add value to the discussion, sorry for that. 

When using "true" as the last parameter, i'm able to scan as often as i want, on iOS. 
On Android i get:
"Callback cannot be null" exception the 2nd time scanning. 

I will look at my Code again, seems like the subscription doesn't get disposed correctly or the StopScan method doesn't get called. In any case the problem seems to be on my side. 
 
Thank you for your time!

aritchie, 2018-04-24:
StopScan was meant to be an emergency brake.  If you call it, existing scans will likely implode like you are seeing.  This is why I default ScanExtra(restart) to false.

I'll take a peak at the android side, but not sure anything will materialize from it