aritchie/bluetoothle #174: NuGet v6.0 missing Plugin.BluetoothLE.dll & v5.3.5 connecting issue

ren0d1, 2018-04-25:
I tried to use the v6.0, but when I launch my app it says that the lib dll is missing.
I then tried to install the v5.3.5, but when I get my scannedDevices and call the device.Connect() it doesn't do a thing nor throw an error.

I also tried to replicate your samples, but nothing is working. I'm probably doing something wrong, but it would be nice to get some hints! ;)

Kind regards,
ren0d1

aritchie, 2018-04-25:
Please use the issue template!!!  What platform is giving you trouble?  Did you make sure to install the nuget package in your android & ios projects as well as you .netstandard project?

ren0d1, 2018-04-26:
## Version of OS(s) listed above with issue
I've only tried to use the shared code part for an android device, since I try one OS at a time;
 - [] Android
 - [] .NET Standard

## Version of Library
v.6 installed in .netstandard, android & ios projects. (At least NuGet manager says so)

## Expected Behaviour
using Plugin.BluetoothLE then use plugin and run fine.

## Actual Behavior
System.ArgumentException: [Plugin.BluetoothLE] No platform plugin found. Did you install the nuget package in your android project as well?
--------------------------------------------------------------------------------------------------------
## Version of OS(s) listed above with issue
I've only tried to use the shared code part for an android device, since I try one OS at a time;
 - [] Android
 - [] .NET Standard

## Version of Library
v.5.3.5 installed in .netstandard, android & ios projects.

## Expected Behaviour
Scan devices -> connect to scanned devices -> communicate

## Actual Behavior
CrossBleAdapter.Current.Scan() -> 2 devices found -> device.Connect() -> CrossBleAdapter.Current.GetConnectedDevices() -> no device connected

P.S.: Just found out that I doing the following works, it was just not stated in your doc
device.Connect().Subscribe( () => {
   List<IDevice> connected = new LIst<IDevice>(CrossBleAdapter.Current.GetConnectedDevices());
   // OR EVEN AN EMPTY SUBSCRIBE WORKS
});

aritchie, 2018-04-26:
The docs you're seeing are for v6.  v6 - Connect is a void, v5 - Connect is an Observable.  I don't always doc that you need to subscribe to observables because it is part of RX 101.  You really need to have a thorough understanding of RX if you are going to have any luck with this library.

Ok - Please don't complain about docs.  I'm 1 guy offering a free lib.  I consider the 5.3.5 issue a non-issue, so let's focus on the v6 as it is really the only thing I offer support on at this point.

For the v6 issue, what's your SDK compile target in Android?  This lib uses 8.  Sometimes nuget is an issue with this.  Verify the version that is installed in your projects as well.  You can also try uninstall/reinstall as well as deleting your bin/obj folders to clear out the Xamarin stuff.

 If it still isn't working for you after that, you will need to submit a reproducible sample.

ren0d1, 2018-04-26:
Oh okay, that's what got me confused.. 
I tried to force min sdk to be 18 for android and reinstalling while following the steps you gave me, but the error is still there. (I simply create new xamarin solution, install the nuget with the nuget manager in .net, android and ios projects and then get the error when I build).
I'll then use the v5.3.5 since it's working if I keep in mind to use observables properly! ahah

Btw, I'm sorry if you misunderstood my intentions.. I didn't want to be rude, I find your work awesome and your doc well written. I even switched from another library to yours for your awesome support and maintenance + the effectiveness of your lib! I just forgot to look at the older doc which I'm sure was containing the subscription part.

aritchie, 2018-04-27:
Thank you for the kind words. 

I'm glad you've got 5.3.5 working, but you will encounter issues fighting with android on that version.  v6 has vast improvements in dealing with android properly.  

That being said, I would like to get to the bottom of what the problem is.  If you could submit a sample of the issue and/or answer a few questions, that would be helpful

1) are you using VS for Mac or Windows?
2) does your android project have a packages.config file?
3) after installing v6 in your android project, can you verify that v6 was indeed installed.
4) For android, it isn't the min sdk you need to set, it's the "Target Android version".  Please ensure SDK 26 or 27 are set.

closed 2018-05-08 by aritchie

prasadsunny1, 2019-07-04:
Still facing the issue with plugin installed in all the projects
`  at Plugin.BluetoothLE.CrossBleAdapter.get_Current () [0x00007] in C:\dev\acr\bluetoothle\Plugin.BluetoothLE\CrossBleAdapter.cs:28 
  at MyCoolProject.Bluetooth.Device.AdapterStatus.CheckBluetoothStatus () [0x00007] in /Users/sunny/Documents/Projects/MyCoolProject.Bluetooth.Tester/MyCoolProject.Bluetooth.Device/AdapterStatus.cs:15 
  at MyCoolProject.Bluetooth.Tester.MainPage.CheckBluetoothAdapter () [0x0000f] in /Users/sunny/Documents/Projects/MyCoolProject.Bluetooth.Tester/MyCoolProject.Bluetooth.Tester/MainPage.xaml.cs:99 
  at MyCoolProject.Bluetooth.Tester.MainPage.BtnServer_Clicked (System.Object sender, System.EventArgs e) [0x00020] in /Users/sunny/Documents/Projects/MyCoolProject.Bluetooth.Tester/MyCoolProject.Bluetooth.Tester/MainPage.xaml.cs:70 
  at System.Runtime.CompilerServices.AsyncMethodBuilderCore+<>c.<ThrowAsync>b__6_0 (System.Object state) [0x00000] in <3a61d48d66dd458fa0b64990b6c8f8cc>:0 
  at Android.App.SyncContext+<>c__DisplayClass2_0.<Post>b__0 () [0x00000] in <ff9f3f3d4e134974a889db1b532c9b6e>:0 
  at Java.Lang.Thread+RunnableImplementor.Run () [0x00008] in <ff9f3f3d4e134974a889db1b532c9b6e>:0 
  at Java.Lang.IRunnableInvoker.n_Run (System.IntPtr jnienv, System.IntPtr native__this) [0x00009] in <ff9f3f3d4e134974a889db1b532c9b6e>:0 
  at (wrapper dynamic-method) Android.Runtime.DynamicMethodNameCounter.27(intptr,intptr)`

aritchie, 2019-07-04:
@prasadsunny1  You need to submit a reproducible sample as a new issue with details.  Please don't hijack old issues.

vranap, 2019-07-17:
Hello, I have the same problem with plugin installed in all the projects. Is that an old problem? Where can I find the answer? Thanks a lot.