aritchie/bluetoothle #175: Device.CancelConnection(); Crash

snowppy, 2018-04-26:
## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [] iOS
 - [*] Android
 - [] tvOS
 - [] macOS
 - [] .NET Standard
 - UWP IS NOT SUPPORTED DURING THE BETA PERIOD - ONLY PRs ARE WELCOME FOR NOW

## Version of OS(s) listed above with issue

Android 6.0.1 and Android version 7

## Version of Library
Latest stable 6.0.0

## Expected Behaviour
device.CancelConnection();
disconnects the device

## Actual Behavior
Just after the call executes the following two exception  are thrown: 
`GattDescriptor+<>c__DisplayClass6_1.<WriteInternal>b__2 ()
DeviceContext+<>c__DisplayClass17_0.<InvokeOnMainThread>b__0 (System.Object _)
SyncContext+<>c__DisplayClass2_0.<Post>b__0 ()
Thread+RunnableImplementor.Run ()
IRunnableInvoker.n_Run (System.IntPtr jnienv, System.IntPtr native__this)
(wrapper dynamic-method) System.Object:c24b57f7-4565-493d-8835-f756b3d3359e (intptr,intptr)`

or 

`
GattCharacteristic+<<DisableNotifications>b__10_0>d.MoveNext ()
ExceptionDispatchInfo.Throw ()
TaskAwaiter.ThrowForNonSuccess (System.Threading.Tasks.Task task)
TaskAwaiter.HandleNonSuccessAndDebuggerNotification (System.Threading.Tasks.Task task)
TaskAwaiter.ValidateEnd (System.Threading.Tasks.Task task)
ConfiguredTaskAwaitable`1+ConfiguredTaskAwaiter[TResult].GetResult ()
DeviceContext+<>c__DisplayClass15_1`1+<<Invoke>b__1>d[T].MoveNext ()
ExceptionDispatchInfo.Throw ()
ExceptionServicesImpl.Rethrow (System.Exception exception)
ExceptionHelpers.Throw (System.Exception exception)
<.cctor>b__2_1 (System.Exception ex)
AnonymousObserver`1[T].OnErrorCore (System.Exception error)
ObserverBase`1[T].OnError (System.Exception error)
AutoDetachObserver`1[T].OnErrorCore (System.Exception exception)
ObserverBase`1[T].OnError (System.Exception error)
DeviceContext+<>c__DisplayClass15_1`1+<<Invoke>b__1>d[T].MoveNext ()
ExceptionDispatchInfo.Throw ()
TaskAwaiter.ThrowForNonSuccess (System.Threading.Tasks.Task task)
TaskAwaiter.HandleNonSuccessAndDebuggerNotification (System.Threading.Tasks.Task task)
TaskAwaiter.ValidateEnd (System.Threading.Tasks.Task task)
TaskAwaiter.GetResult ()
DeviceContext+<ProcessQueue>d__20.MoveNext ()
ExceptionDispatchInfo.Throw ()
AsyncMethodBuilderCore+<>c.<ThrowAsync>b__6_0 (System.Object state)
SyncContext+<>c__DisplayClass2_0.<Post>b__0 ()
Thread+RunnableImplementor.Run ()
IRunnableInvoker.n_Run (System.IntPtr jnienv, System.IntPtr native__this)
(wrapper dynamic-method) System.Object:4802edd7-24f1-4623-abcd-ad80d03eef90 (intptr,intptr)
`

## Code of Conduct
* No Library Update Requests
* No Timeline Requests
* Please read through the documentation and FAQs
* Please be respectful of my time. I put a lot of effort into my libraries.  I try to support the issue requests within a day.  I'm not psychic, so please take the time to fill out the above form as complete as possible

aritchie, 2018-04-26:
Are you using Characteristic.RegisterAndNotify?

snowppy, 2018-04-26:
I do have a read and Notify on a  read characteristic
`
  readCharacteristicObserver = characteristic.RegisterAndNotify(true).Subscribe(r =>
                            {
                                Read?.Invoke(r.Data);
                            });        `
I thought I was disposing of the listener correctly since it was working correctly on the previous version of the package.
this is the disconnect code:
` if (readCharacteristic != null) {
                    try
                    {
                        readCharacteristicObserver.Dispose();
                        device.CancelConnection();
                    }
                    catch (Exception e)
                    {
                        Crashes.TrackError(e);
                    }
                }`

aritchie, 2018-04-26:
The bug isn't yours.  However, you shouldn't treat that exception as a huge problem.  You should always hook to the onError portion of the subscribe in any case especially since RegisterAndNotify does go out to a write operation that can fail even when a connection is present.  That isn't a bug.  It is simply the nature of the beast.  You can usually treat them as "catch & release" and retry the operation later.

In this case, the bug is likely due to the way I'm cleaning up the RegisterAndNotify which I already have a test and fix for.  I just need to verify sometime in the next day or two.

You can work around this by simply doing Characteristic.EnableNotifications() and hooking Characteristic.WhenNotNotificationReceived yourself.