aritchie/bluetoothle #180: Write characteristic not writing after reconnect

snowppy, 2018-04-30:
## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [] iOS
 - [*] Android
 - [] tvOS
 - [] macOS
 - *] .NET Standard
 - UWP IS NOT SUPPORTED DURING THE BETA PERIOD - ONLY PRs ARE WELCOME FOR NOW

## Version of OS(s) listed above with issue
6.0.1

## Version of Library
6.0.2

## Expected Behaviour
Current scenario
1. Get Paired devices,
2 Connect to paired device
3 WhenAnyCharacteristicDiscovered returns Expected Read and Write characteristics
4. Subscribe on the read characteristic RegisterAndNotify(true)
5. Write a few payloads with the write characteristic .Write(pload)
6. Expect write characteristic to respond either GattResult or on OnError exception 
7. Disconnect device and dispose all observables.
8 Repeat steps 1-7 and get same behavior.

## Actual Behavior
On the first connect everything is working as expected. all the 7 steps above execute as expected.
On disconnect I dispose all the observables in use ( WhenAnyCharacteristicDiscovered  and RegisterAndNotify(true)).
I remove all references for the device or any characteristic.
I start repeating the the steps described above.
I should point out that step 3 returns the result faster than the first run.
on Step 5 when I write a payload to the characteristic I get stuck. Neither GatResult or OnError is executed and I am stuck In case of waiting the write characteristic to respond.

`    var obs=  writeCharacteristic.Write(pload).Subscribe(result =>
                            {
                                writeResult = true;
                            },
                            exception =>
                            {
                                Crashes.TrackError(exception);
                                writeResult = false;
                            });`

## Code of Conduct
* No Library Update Requests
* No Timeline Requests
* Please read through the documentation and FAQs
* Please be respectful of my time. I put a lot of effort into my libraries.  I try to support the issue requests within a day.  I'm not psychic, so please take the time to fill out the above form as complete as possible

aritchie, 2018-05-01:
Please submit a small reproducible sample in the future.  Reiterating a one line command without the rest of it isn't helpful.  Also, please read the general rules for working with BLE.  You should be using timeouts in reads/writes.  

In any case, I'm already working on this one internally.

snowppy, 2018-05-01:
Hi, I made a small sample of how I have it setup that recreates the behavior from above [here](https://github.com/snowppy/BleSamples).

aritchie, 2018-05-07:
This definitely didn't constitute a small repo.  I've made changes in 6.0.3 with another one coming in 6.0.4 that should deal with this.

aritchie, 2018-06-22:
Fixed in upcoming 6.1

closed 2018-06-22 by aritchie