aritchie/bluetoothle #183: Major performance loss for Send/NotificationRecieved between 6.0.0-beta3 and 6.0.2/6.0.3

borbmizzet, 2018-05-07:
## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [] iOS
 - [*] Android
 - [] tvOS
 - [] macOS
 - [*] .NET Standard
 - UWP IS NOT SUPPORTED DURING THE BETA PERIOD - ONLY PRs ARE WELCOME FOR NOW

## Version of OS(s) listed above with issue

Android 7.0, Galaxy S6 Edge+
## Version of Library
6.0.2/6.0.3 (tested with both)

## Expected Behaviour
No performance loss when upgrading from 6.0.0-beta3 to 6.0.2/6.0.3

## Actual Behavior
IGattCharacteristic.Write and/or NotificationRecieved observable (not sure which, possibly both) take around 2x as long after upgrading from 6.0.0-beta3 to 6.0.2.  Performance hit remains in 6.0.3.  Tested with clean builds between 2 consecutive commits where the only change was the Plugins.BluetoothLE package update.

## Code of Conduct
* No Library Update Requests
* No Timeline Requests
* Please read through the documentation and FAQs
* Please be respectful of my time. I put a lot of effort into my libraries.  I try to support the issue requests within a day.  I'm not psychic, so please take the time to fill out the above form as complete as possible

aritchie, 2018-05-07:
Do you have a SMALL test case or any sort of numbers showing this?  I'll be honest, I'm not truly worried about performance since the write is within a few 300-400 ms in my tests which is acceptable.  There is also a fair bit of difference between beta and release, so you don't need to quote it here.

You can use CrossBleAdapter.PauseBetweenInvocations set to null to speed things up as well.

aritchie, 2018-05-07:
The times are acceptable.  My goal with v6 was to stop the user from overwhelming the device.  Internally, I'm using a processing queue which can slow things down for reads & writes (notification rx has not changed and does not flow through the queue).  I also have a pause of 200ms between each operation as stated above.  You can set that to NULL and try your tests (In android project CrossBleAdapter.PauseBetweenInvocations = null; ).

Aside from that, unless you can find something that pinpoints the issue, I'm likely going to leave this.

borbmizzet, 2018-05-07:
That fixed the issue.  Are there any pitfalls I should watch out for after doing that?

aritchie, 2018-05-07:
Shitty devices (LG, older samsung devices, etc) can potentially be overwhelmed.  The queue mechanism seems to be working well overall to prevent this though.

closed 2018-05-07 by aritchie