aritchie/bluetoothle #184: Android/iOS Mismatch Response on BlobWrite, what am I doing wrong?

Uniphix, 2018-05-07:
## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [x] iOS
 - [x] Android
 - [] tvOS
 - [] macOS
 - [] .NET Standard
 - UWP IS NOT SUPPORTED DURING THE BETA PERIOD - ONLY PRs ARE WELCOME FOR NOW

## Version of OS(s) listed above with issue
API 25 Android, iOS 11+

## Version of Library
6.0.3

## Expected Behaviour
Write to the Device

## Actual Behavior
https://github.com/aritchie/bluetoothle/blob/master/Plugin.BluetoothLE/AbstractGattCharacteristic.cs#L67
Got an actual mismatch exception not sure what this is

## Code of Conduct
* No Library Update Requests
* No Timeline Requests
* Please read through the documentation and FAQs
* Please be respectful of my time. I put a lot of effort into my libraries.  I try to support the issue requests within a day.  I'm not psychic, so please take the time to fill out the above form as complete as possible

aritchie, 2018-05-08:
It's a left over artifact that will be removed in 6.0.4

closed 2018-05-08 by aritchie

Uniphix, 2018-05-08:
Is there a work around for this right now?  I know it will be removed in 6.0.4