aritchie/bluetoothle #186: BLE device connect status always Disconnected.

LeonanFragaLeonardo, 2018-05-15:
## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [] iOS
 - [X] Android
 - [] tvOS
 - [] macOS
 - [] .NET Standard
 - UWP IS NOT SUPPORTED DURING THE BETA PERIOD - ONLY PRs ARE WELCOME FOR NOW

## Version of OS(s) listed above with issue
Android 6.0

## Version of Library
5.3.5

## Expected Behaviour
Connect a Device.

## Actual Behavior
There is no exception, but when I do this: "device.Connect();", the device status has not changed.
Code:
I have a ListView which contains all discovered BLE devices as IScanResult.
Here is my Code:
`private void OnItemTapped(object o, ItemTappedEventArgs args)
{
   var x = args.Item as IScanResult;
   x.Device.Connect();
}`

But, even if I do this, the connection status keep Disconnected.

LeonanFragaLeonardo, 2018-05-15:
Yes, my problem is exactly this: 
https://github.com/aritchie/bluetoothle/issues/174
However, there are no answer related to this problem.
Here is my Android packages.config
`<package>
...
  <package id="Plugin.BluetoothLE" version="6.0.3" targetFramework="monoandroid71" />
...
</package>`

and my Manifest.xml

`<?xml version="1.0" encoding="utf-8"?>
<manifest xmlns:android="http://schemas.android.com/apk/res/android" android:installLocation="auto">
	<uses-sdk android:minSdkVersion="15" android:targetSdkVersion="26" />
	<uses-permission android:name="android.permission.BLUETOOTH" />
	<uses-permission android:name="android.permission.BLUETOOTH_ADMIN" />
	<uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" />
	<uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION" />
	<uses-permission android:name="android.permission.BLUETOOTH_PRIVILEGED" />
	<uses-feature android:name="android.hardware.bluetooth_le" android:required="true" />
	<application android:label="Tracking.Android"></application>
</manifest>`

Thank you for your help.
Sorry for my bad English.

aritchie, 2018-05-15:
Before with Connect, you had to subscribe because it was an observable.  The Connect subscription is async and will not block while waiting to connect

aritchie, 2018-05-15:
What install problems?  I’m not psychic... you have to give me details if you want help

LeonanFragaLeonardo, 2018-05-15:
The package installs correctly, but when some BLE code is executed, for instance:
`scan = CrossBleAdapter.Current.Scan().Subscribe(scanResult =>
                {
                    scanResults.Add(scanResult);
                });
`
the VS throws this error:

"**System.ArgumentException: [Plugin.BluetoothLE] No platform plugin found. Did you install the nuget package in your android project as well?**"

My target framework is**.NET standard 2.0.** and my **Xamarin.Forms** is 3.0.0.482510

Sorry if I have not detailed my problem before. 😞

LeonanFragaLeonardo, 2018-05-16:
I'm here again, but now to show how to I solved my problem.
(Just for PCL Project): 
 * Upgrade the project from PCL to .NetStandard2.0. [Link to Montemagno tutorial](https://montemagno.com/how-to-convert-a-pcl-library-to-net-standard-and-keep-git-history/). Ps: **Don't forget to remove the packages.config and AssemblyInfo.cs**

 * After to upgrade your project, uninstall "**Microsoft.NETCore.Portable.Compatibility**." installed in your project, if applicable. 

* Download, install and do the conversion using the [NugetPackageConfigConverter VS Extension](https://github.com/twsouthwick/NuGetPackageConfigConverter).

That's all. Now your PCL project is able to install the latest version (v6) of the BluetoothLE plugin. I hope that I can help someone with this solution.