aritchie/bluetoothle #195: Raspberry PI 3 B and Bluetooth

Naografix, 2018-06-11:
## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [x] iOS
 - [x] Android 
 - [x] .NET Standard 
 - [x] Xamarin Forms 

## Version of Library
v6.0.3

I tried to scan my network to discovered my rpi 3 with your Sample and I can't see it in the list and I don't understand why.
It's weird because I can see the rpi when I scan with my phone in my Bluetooth settings, but not with your sample.

Thanks for your help

aritchie, 2018-06-11:
This library supports Bluetooth LE, not standard bluetooth.  You need to make sure that your device is broadcasting on BLE.

Naografix, 2018-06-11:
@aritchie Thank you.
I need to know how to broadcast on BLE now, because I can see here, RPI 3 have BLE:
https://www.raspberrypi.org/magpi/raspberry-pi-3-specs-benchmarks/

Have you got any idea how to broadcast on BLE?

aritchie, 2018-06-11:
I have no idea about the external device.  I know this library scans successfully for BluetoothLE devices.

closed 2018-06-12 by aritchie