aritchie/bluetoothle #196: Android: Enable Notification Error

nviswanath24, 2018-06-15:
Hi,

I'm just using your sample code 'Samples'. When I try to enable the notifications in a characteristic, I get the error below. The example works fine with iOS, I only get the error with Android. (I have tried to test with the SetCharacteristicNotification on Android studio and it works fine with my BLE device. )

## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)
 - [x] Android

## Version of Library
6.0.3

## Expected Behaviour
Notification should be enabled successfully

## Actual Behavior
"Characteristic Client Configuration Descriptor not found" Plugin.BluetoothLE.BleException

Thanks!

nviswanath24, 2018-06-15:
The characteristic properties are write and notify. When I tap the characteristic ->tap notify -> HEX/UTF8, the app crashes and the above exception message is displayed. Thanks!

aritchie, 2018-06-15:
do you have a description of the device?  Maybe from another app?  I have not observed the problem before.

nviswanath24, 2018-06-15:
Unfortunately I don't. But the app works fine on iOS and I'm only using your sample code. 

 async void ToggleNotify()
        {
            if (this.Characteristic.IsNotifying)
            {
                this.watcher?.Dispose();
                this.IsNotifying = false;
            }
            else
            {
                this.IsNotifying = true;
                var utf8 = await UserDialogs.Instance.ConfirmAsync(
                    "Display Value as UTF8 or HEX?",
                    okText: "UTF8",
                    cancelText: "HEX"
                );
                await this.Characteristic.EnableNotifications(true);
                this.watcher = this.Characteristic
                    .WhenNotificationReceived()
                    .Subscribe(x => UserDialogs.Instance.ConfirmAsync(utf8
                    ? Encoding.UTF8.GetString(x.Data, 0, x.Data.Length)
                                                    : BitConverter.ToString(x.Data))); 
            }
        }

It fails at the EnableNotifications line on Android.

aritchie, 2018-06-15:
Can you get one?  I’m not worried about the sample.  I want to figure our why your host device is being picked up with a characteristic with notify properties that doesn’t appear to have it.

What version of android are you running and what type of device?

nviswanath24, 2018-06-15:
I'm running Android 8.1 on a Google Pixel 2. The notification works perfectly on iOS.

nviswanath24, 2018-06-15:
The same service also has a read, notify characteristic. When I try to enable notification on that it returns this error message - Failed to write descriptor value - WriteNotPermitted

aritchie, 2018-06-15:
Ok please supply a screenshot from a different app and from the sample.  I can’t help you if I can’t see it.  I don’t have access to your host device.

nviswanath24, 2018-06-19:
Is there any way that I can edit the plugin and use it only for my project? I want to test and see if the notification works if I use only 'SetCharacteristicNotification' rather than checking for the descriptor. When I just use SetCharacteristicNotification it works on Android Studio. Any indication of how I could proceed would be very helpful. Thanks!

aritchie, 2018-06-19:
It's open source, you can do whatever you want with it

nviswanath24, 2018-06-19:
Okay, thank you.

closed 2018-06-20 by aritchie