aritchie/bluetoothle #198: CrossBleAdapter.Current throw TypeInitializationException on Android 4.4

priolo22, 2018-06-21:
## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [] iOS
 - [x] Android
 - [] tvOS
 - [] macOS
 - [] .NET Standard
 - UWP IS NOT SUPPORTED DURING THE BETA PERIOD - ONLY PRs ARE WELCOME FOR NOW

## Version of OS(s) listed above with issue
4.4

## Version of Library
6.0.3

## Expected Behaviour
after initializing the view
CrossBleAdapter.Current
give no error:
Unhandled Exception:
System.TypeInitializationException: The type initializer for 'Plugin.BluetoothLE.CrossBleAdapter' threw an exception.

## Actual Behavior
after initializing the view
CrossBleAdapter.Current
give error:
Unhandled Exception:
System.TypeInitializationException: The type initializer for 'Plugin.BluetoothLE.CrossBleAdapter' threw an exception.

## Code of Conduct
```
namespace App1
{
	public partial class MainPage : ContentPage
	{
		public MainPage()
		{
			InitializeComponent();
                        // newxt line trow error: System.TypeInitializationException
                        var c = CrossBleAdapter.Current;
		}
	}
}
```
## Description  

First of all, congratulations for your library! In my opinion very well done.  
Unfortunately I have this problem.  
  
- On a new empty project Xamarin (Shared Project)
- I install the plugin Plugin.BluetoothLE v6.0.3
- Use "CrossBleAdapter.Current"
- Launch debug on a real Android **4.4** device
- Throw error: System.TypeInitializationException
**on android> 4.4 it works**
  
I suppose I ignore something!  
Thank you

aritchie, 2018-06-21:
Please make sure to get include the entire stacktrace in the future. 

My guess is that you are getting “platform plugin not installed”.  Make sure your android target is compiled with v8 then reinstall plugin.  Also make sure you aren’t using the old Nuget system with packages.config

priolo22, 2018-06-21:
Thanks for the reply!  

- package management: 
PackageReference

- Manifest
```xml
<?xml version="1.0" encoding="utf-8"?>
<manifest xmlns:android="http://schemas.android.com/apk/res/android" android:versionCode="1" android:versionName="1.0" package="com.companyname.App1" android:installLocation="auto">
	<uses-sdk android:minSdkVersion="19" android:targetSdkVersion="27" />
	<uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION" />
	<uses-permission android:name="android.permission.BLUETOOTH" />
	<uses-permission android:name="android.permission.BLUETOOTH_ADMIN" />
	<uses-permission android:name="android.permission.BLUETOOTH_PRIVILEGED" />
	<application android:label="App1.Android"></application>
</manifest>
```

- screenshot error:
[image]

- stacktrace:
```
System.TypeInitializationException: The type initializer for 'Plugin.BluetoothLE.CrossBleAdapter' threw an exception. ---> Java.Lang.NoClassDefFoundError: md5a39dd0081ac4a79a679ecc2b304d9848/AdvertisementCallbacks ---> Java.Lang.ClassNotFoundException: Didn't find class "md5a39dd0081ac4a79a679ecc2b304d9848.AdvertisementCallbacks" on path: DexPathList[[zip file "/data/app/com.companyname.App1-1.apk"],nativeLibraryDirectories=[/data/app-lib/com.companyname.App1-1, /vendor/lib, /system/lib]]
   --- End of inner exception stack trace ---
  at Java.Interop.JniEnvironment+Types.FindClass (System.String classname) [0x00114] in <7802aa64ad574c33adca332a3fa9706a>:0 
  at Java.Interop.JniType..ctor (System.String classname) [0x00006] in <7802aa64ad574c33adca332a3fa9706a>:0 
  at Java.Interop.JniPeerMembers+JniInstanceMethods..ctor (System.Type declaringType) [0x00064] in <7802aa64ad574c33adca332a3fa9706a>:0 
  at Java.Interop.JniPeerMembers+JniInstanceMethods.GetConstructorsForType (System.Type declaringType) [0x0002c] in <7802aa64ad574c33adca332a3fa9706a>:0 
  at Java.Interop.JniPeerMembers+JniInstanceMethods.StartCreateInstance (System.String constructorSignature, System.Type declaringType, Java.Interop.JniArgumentValue* parameters) [0x00032] in <7802aa64ad574c33adca332a3fa9706a>:0 
  at Android.Bluetooth.LE.AdvertiseCallback..ctor () [0x00034] in <263adecfa58f4c449f1ff56156d886fd>:0 
  at Plugin.BluetoothLE.Server.Internals.AdvertisementCallbacks..ctor () [0x00000] in <5b158e307d7547c0ac6875db9ce3d1b5>:0 
  at Plugin.BluetoothLE.Server.Advertiser..ctor () [0x00020] in <5b158e307d7547c0ac6875db9ce3d1b5>:0 
  at Plugin.BluetoothLE.Adapter..ctor () [0x00031] in <5b158e307d7547c0ac6875db9ce3d1b5>:0 
  at Plugin.BluetoothLE.CrossBleAdapter..cctor () [0x00042] in <5b158e307d7547c0ac6875db9ce3d1b5>:0 
   --- End of inner exception stack trace ---
  at App1.MainPage..ctor () [0x00010] in C:\Users\Ivano\Desktop\testbluetootsu4_4\App1\App1\MainPage.xaml.cs:17
```

aritchie, 2018-06-21:
Looks like a defect with older android.

priolo22, 2018-06-21:
Ok, thanks for your time.   
I appreciate it very much!

priolo22, 2018-06-21:
may I know if you plan to fix that in the very next future?
Thank you

closed 2018-06-22 by aritchie

priolo22, 2018-06-22:
You are right. This weekend I will give an eye and I will let you know!
However I suspect it is a general thing of xamarin
Meanwhile, thank you for everything

commit 653dcd80 (2018-06-22): fix #198 - check for API level 21 on advertising, also didn't need to be instantiated out of the gate