aritchie/bluetoothle #199: no devices found

ipon85, 2018-06-21:
## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [] iOS
 - [x] Android
 - [] tvOS
 - [] macOS
 - [] .NET Standard
 - UWP IS NOT SUPPORTED DURING THE BETA PERIOD - ONLY PRs ARE WELCOME FOR NOW

## Version of OS(s) listed above with issue
8.0

## Version of Library
5.0, 6.0.3

## Expected Behaviour
found BLE devices

## Actual Behavior
no BLE devices found on android 8.0 (on LG G6). On android 7.0 all was ok

## Code of Conduct
* No Library Update Requests
* No Timeline Requests
* Please read through the documentation and FAQs
* Please be respectful of my time. I put a lot of effort into my libraries.  I try to support the issue requests within a day.  I'm not psychic, so please take the time to fill out the above form as complete as possible

aritchie, 2018-06-21:
Please provider a small, end-to-end, reproducible sample.  I run my ble stuff on android 7.1 & 8 all the time without issue.

ipon85, 2018-06-21:
well, seems it somehow uncheck permissions ACCESS_COARSE_LOCATION and ACCESS_FINE_LOCATION. In debug:

`06-21 17:45:53.704 D/BluetoothLeScanner(14451): onScannerRegistered() - status=0 scannerId=5 mScannerId=0
06-21 17:45:53.725 W/Binder  (14451): Caught a RuntimeException from the binder stub implementation.
06-21 17:45:53.725 W/Binder  (14451): java.lang.SecurityException: Need ACCESS_COARSE_LOCATION or ACCESS_FINE_LOCATION permission to get scan results
06-21 17:45:53.725 W/Binder  (14451): 	at android.os.Parcel.readException(Parcel.java:1942)
06-21 17:45:53.725 W/Binder  (14451): 	at android.os.Parcel.readException(Parcel.java:1888)
06-21 17:45:53.725 W/Binder  (14451): 	at android.bluetooth.IBluetoothGatt$Stub$Proxy.startScan(IBluetoothGatt.java:976)
06-21 17:45:53.725 W/Binder  (14451): 	at android.bluetooth.le.BluetoothLeScanner$BleScanCallbackWrapper.onScannerRegistered(BluetoothLeScanner.java:456)
06-21 17:45:53.725 W/Binder  (14451): 	at android.bluetooth.le.IScannerCallback$Stub.onTransact(IScannerCallback.java:56)
06-21 17:45:53.725 W/Binder  (14451): 	at android.os.Binder.execTransact(Binder.java:690)`

but permissions are present 
`	<uses-permission android:name="com.google.android.c2dm.permission.RECEIVE" />
	<uses-permission android:name="android.permission.WAKE_LOCK" />
	<uses-permission android:name="android.permission.INTERNET" />
	<uses-permission android:name="com.flypenny.android.permission.C2D_MESSAGE" />
	<uses-permission android:name="android.permission.ACCESS_NETWORK_STATE" />
	<uses-permission android:name="android.permission.ACCESS_WIFI_STATE" />
	<uses-permission android:name="android.permission.READ_CONTACTS" />
	<uses-permission android:name="android.permission.WRITE_EXTERNAL_STORAGE" />
	<uses-permission android:name="android.permission.READ_EXTERNAL_STORAGE" />
	<uses-permission android:name="android.permission.BLUETOOTH_ADMIN" />
	<uses-permission android:name="android.permission.BLUETOOTH" />
	<uses-permission android:name="android.permission.BLUETOOTH_PRIVILEGED" />
	<uses-permission android:name="android.permission.READ_SMS" />
	<uses-permission android:name="android.permission.RECEIVE_SMS" />
	<uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION" />
	<uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" />`

forgot to say, the plugin used in background mode via android.service

aritchie, 2018-06-21:
Glad you found your answer

closed 2018-06-21 by aritchie

Adrianotiger, 2018-10-31:
I have the same problem. Even if it is present, in the debug window I got 
`java.lang.SecurityException: Need ACCESS_COARSE_LOCATION or ACCESS_FINE_LOCATION permission to get scan results`
How did you solved it?