aritchie/bluetoothle #205: calling CancelConnection after a WhenConnected event does not cleanly disconnect unless you add a delay

borbmizzet, 2018-06-28:
## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [] iOS
 - [*] Android
 - [] tvOS
 - [] macOS
 - [] .NET Standard
 - UWP IS NOT SUPPORTED DURING THE BETA PERIOD - ONLY PRs ARE WELCOME FOR NOW

## Version of OS(s) listed above with issue
7.0.0

## Version of Library
6.1

## Expected Behaviour
The below code should work without anything at <<LOCATION1>>

## Actual Behavior
`IDevice device;
var dis1 = Adapter.Scan().First().Subscribe(r => device = r.Device);
await Task.Delay(5000);
dis1.Dispose();
TaskCompletionSource<bool> tcs = new TaskCompletionSource<bool>()
device.WhenConnected().Subscribe(() => {tcs.TrySetResult(true);});
device.Connect(new ConnectionConfig{AutoConnect = false});
await tcs.Task;
<<LOCATION1>>>
Device.Disconnect();
IDevice device2;
var dis2 = Adapter.Scan().Where(r => device.Uuid == r.Device.Uuid)
.First()
.Subscribe(r=>device2 = r.Device);
await Task.Delay(5000);
dis2.Dispose();
Assert.NotNull(device2);
`
The assert at the end throws unless `await Task.Delay(750);` is placed at `<<LOCATION1>>`

I know this is not a realistic scenario for production but it is realisic for unit and integration testing of an IO layer.
## Code of Conduct
* No Library Update Requests
* No Timeline Requests
* Please read through the documentation and FAQs
* Please be respectful of my time. I put a lot of effort into my libraries.  I try to support the issue requests within a day.  I'm not psychic, so please take the time to fill out the above form as complete as possible

aritchie, 2018-06-28:
Please submit a small compilable reproducible sample showing this issue.  What is your definition of “cleanly disconnecting”.  Is it throwing an exceptional?  If so, what’s the exception?

borbmizzet, 2018-06-29:
The key there is the `Assert.NotNull(device2)` at the end.  The assertion is essentially asserting whether the peripheral device represented by `device` started advertising again after `device.CancelConnection()`.  The assertion fails unless there is a delay between observing `device.WhenConnected()` and calling `device.CancelConnection()`.  With no delay, the peripheral device does not resume advertising after supposedly disconnecting, even after waiting multiple minutes.  I've verified this with a sniffer.

aritchie, 2018-06-29:
Cancelling a connection is technically an async process.  I don't plan on showing that since it really isn't a necessary case and I have all of the other tools in place to watch for things like this.

You can do device.CancelConnection();  await Device.WhenDisconnected().Take(1).ToTask(); 

This shows an example of how I test reconnections:
https://github.com/aritchie/bluetoothle/blob/master/Plugin.BluetoothLE.Tests/DeviceTests.cs#L168

In the unit tests, I do this case all the time with cancelconnection

closed 2018-06-29 by aritchie

borbmizzet, 2018-06-29:
@aritchie The above code was written as a demonstration.
Remove the code
```
            IDevice device2 = null;
            var dis2 = CrossBleAdapter.Current.Scan().Where(r => device.Uuid == r.Device.Uuid)
                .Take(1)
                .Subscribe(r => device2 = r.Device);
            Task.Delay(5000).Wait();
            dis2.Dispose();
            Debug.Assert(device2 != null);
```
And replace with some arbitrary BLE sniffing procedure some random amount of time after the program ends.  The point is that the peripheral does not advertise at all after disconnect without a delay after the connect.

aritchie, 2018-06-29:
Here are some general suggestion as I don't consider this an issue (and it is edge in any case).

1) You need to await a disconnect status event if you are interested in watching that.  You can use a delay, but it isn't ideal.  
2) Task.Delay with Wait is going to lock the thread and android uses the main thread to do a majority of writes.  Not an issue here, just something to be aware of.
3) You also shouldn't rescan as you'll get a different instance of the device underneath the hood.

Best of luck