aritchie/bluetoothle #211: WhenDisconnected not waiting correctly after CancelConnection

gunpuz, 2018-07-12:
## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [] iOS
 - [*] Android
 - [] tvOS
 - [] macOS
 - [] .NET Standard
 - UWP IS NOT SUPPORTED DURING THE BETA PERIOD - ONLY PRs ARE WELCOME FOR NOW

## Version of OS(s) listed above with issue

## Version of Library
6.1.2

## Expected Behaviour

            Device.DeviceObject.Connect();

            await Device.DeviceObject.ConnectWait();

            Device.DeviceObject.CancelConnection();
            
            await Device.DeviceObject.WhenDisconnected();

"WhenDisconnected()" should wait for device to be at disconnected state
## Actual Behavior
_If applicable, please include entire exception - type, message, stacktrace_

Waits forever?

Intresting note:
status change subscription should be before calling cancel connection?
But if at first cancelConnection is called and then status changes is subscribed, then status is triggered only at "Connected"
This code did the work (Returned after device state is disconnected):

            var tcs = new TaskCompletionSource<bool>();
            Debug.WriteLine(await CrossBleAdapter.Current.GetConnectedDevices().Count());

            IDisposable sub = null;
            sub = Device.DeviceObject.WhenStatusChanged().Subscribe(status =>
            {
                if (status == ConnectionStatus.Disconnected)
                {
                    sub?.Dispose();
                    tcs.TrySetResult(true);
                }
            });
            Device.DeviceObject.CancelConnection();
            
            await tcs.Task;

## Code of Conduct
* No Library Update Requests
* No Timeline Requests
* Please read through the documentation and FAQs
* Please be respectful of my time. I put a lot of effort into my libraries.  I try to support the issue requests within a day.  I'm not psychic, so please take the time to fill out the above form as complete as possible

aritchie, 2018-07-12:
WhenDisconnected is a non-completing observable.  You’ll want to add take(1).  It will fire an initial state so if it is currently disconnected, it will fire.

closed 2018-07-12 by aritchie

gunpuz, 2018-07-13:
Well, you mean something like this? It simply does not cut it, waits forever. Can the reason be what i mentioned about command sequence around CancelConnecion? One other intresting note is, that device is not actually disconnected event after firing disconnected state in my code sample. How to propertly disconnect the device?

            Device.DeviceObject.Connect();
            await Device.DeviceObject.ConnectWait();

            Debug.WriteLine(CrossBleAdapter.Current.GetConnectedDevices().Count());

            Device.DeviceObject.CancelConnection();
            await device.DeviceObject.WhenDisconnected().Take(1);

            Debug.WriteLine(CrossBleAdapter.Current.GetConnectedDevices().Count());

aritchie, 2018-07-13:
Connect/CancelConnection are asynchronous technically under the hood.  I don’t wait them because Connect may connect hours later and cancelconnection may not actually be disconnecting a device, it simply may be cancelling a request for a connection.

I do a very similar chunk of code in my unit tests without issue.

gunpuz, 2018-07-13:
But question remains - if it is disconnecting device or just cancelling connection. Well i would be happy about full disconnect, like "BLE disconnect packet sending disconnect". Because that way i can connect to device from other client.

gunpuz, 2018-07-13:
About Connect wait, like i said, my custom code example for disconnect wait runs smoothly around ~ 1s. Problem might be with command sequence.

aritchie, 2018-07-13:
It is disconnecting the device, it sometimes just takes a moment because it is async.