aritchie/bluetoothle #213: Complex scenario - use cases

Magic73, 2018-07-14:
Hi, 
first of all thanks for your work.

I would like to know if it is possible to have a complete example about how to use it.
I don't mean about the basic operations, but a "complex" scenario like:
- scan for a specific device
- connect to that device
- handle properly autereconnection when turn off BT or turn off BT connected device
- release all resource when shutting down the app

I'm not familiar with RX, and this is another step of complexity.

Have you a complex example?

aritchie, 2018-07-15:
There are samples and unit tests included in this repository that will help you out.  If you don’t know RX, you will have issues using this library though.

There are numerous extension methods off the adapter and device interfaces that will cutdown on the amount of work you have to do.  I would look through the source code to get a decent understanding of it before using it.

closed 2018-07-15 by aritchie

Magic73, 2018-07-15:
I already used this library to develop and publish an app.
the documentation and the samples are you an introduction which cover a single piece.

Example:
I start a scan, I detect a device, I connect to the device. Should I dispose the scanner? If I dispose it, the autoreconnection works?

In the code I'm using above, scanner is faster than device connection. How may I pause it? Should I dispose it just after I found the device, even if I don't know if I'll able to connect ?

what I'm asking is what are "your" best practice to handle these common scenario, since it's your library. you know how to use it, and I shouldn't read and understand each line of code of the library unless it would be necessary (as example, I fixed all the UWP issue, it disconnect correctly.. but I can't push an update, since you changed the library a month ago... btw, one thing useless and which prevent disconnection is the PING..).

Regards,
Cristiano

aritchie, 2018-07-16:
I’m sorry - questions like this, I just don’t have the time or energy to answer.  There are so many things involved with bluetooth and I do try to answer some of them in the docs (some of which answer the questions you are asking).  I try to add to I every now and again as I see people doing something repeatedly off what was intended.

Lastly, I really don’t care for your last part about UWP.  I find it extremely rude in fact.  You’ve mentioned that I’m blocking you from releasing and the fact that you’ve made a bunch of fixes without any contribution back!?!?  Come on...  when you say things like this why would I even attempt to help answer your questions

Magic73, 2018-07-16:
Emh, maybe my english is not so good, but I never say that you are blocking me from releasing.
the app is already released and it  works (search "Magic ARC App" in the 3 stores). 
About fixes, I explained well.. I did them but when I come back to push fixes, You changed the source code a lot. (I download the source code on 8 of March), and I tryed to push it, if I remember well… on May).

anyway, I will find a way to understanad which is the best pattern to do what I need.
Thanks again for your library.

Cristiano