aritchie/bluetoothle #216: How to get specific characteristics sequentially?

diescalo, 2018-07-23:
## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [] iOS
 - [*] Android
 - [] tvOS
 - [] macOS
 - [] .NET Standard
 - UWP IS NOT SUPPORTED DURING THE BETA PERIOD - ONLY PRs ARE WELCOME FOR NOW

## Version of OS(s) listed above with issue
Android phone with API 21

## Version of Library
6.1.3

## Expected Behaviour
I want to get a reference of some characteristics so that I can use later in the app to write values to, but I want to control when they are obtained.

## Actual Behavior
I'm a bit confused about the library because I can't get it working with my device (at least the way I was expecting). So, I'm able to find BLE devices and add them to a list, bu I'm having problems going a bit further. I have read the characteristics documentation but it doesn't explain how to do that without subscribing to an automatic discovery.

For example, how do I wait for the device to be connected? I mean, as soon as I call the `Connect` method, application continues execution without waiting for the device to be fully connected and I can't **await** that method.

Just after connecting I want to get the reference to 2 specific characteristics and subscribe to one of them, but I can't figure out how to do that sequentially. I have tried:

```
private Guid txCharID = Guid.Parse("TX CHAR  UUID");
private Guid rxCharID = Guid.Parse("RX CHAR  UUID");

private byte[] txData = new byte[] { 0x01, 0x23, 0x45, 0x67, 0x89 };

private IGattCharacteristic txCharacteristic = null;
private IGattCharacteristic rxCharacteristic = null;

...

Device.Connect();  // Execution continues without waiting for the device to be connected.

Device.WhenConnected().Subscribe(connectedDevice =>
{
	Debug.WriteLine("----- Device status: " + Device.Status.ToString());
});

Device.WhenAnyCharacteristicDiscovered().Subscribe(characteristic =>
{
	Debug.WriteLine("----- Discovered characteristic > " + characteristic.Uuid);

	if (characteristic.Uuid.Equals(txCharID))
	{
		txCharacteristic = characteristic;
	}
	else if (characteristic.Uuid.Equals(rxCharID))
	{
		rxCharacteristic = characteristic;
		if (rxCharacteristic.CanNotifyOrIndicate())
		{
			rxCharacteristic.EnableNotifications(true);
			rxCharacteristic.WhenNotificationReceived().Subscribe(result =>
			{
				Debug.WriteLine("----- Data received");
			});
			Debug.WriteLine("----- Subscribed to Rx characteristic");
		}
	}
});

// This line is executed before the device is connected and characteristics discovered.

```
But I can't control when characteristics are obtained nor the order they are being discovered.

Maybe I'm totally wrong and the API was not designed to work the way I'm trying to use it, but other APIs I tried had clear ways to do what I'm trying to achieve.

Thanks!

PS: I tried to follow the documentation, but even the connection part is not working for me: https://github.com/aritchie/bluetoothle/blob/master/docs/device.md I get errors just copy/pasting the code displayed in that page.

## Code of Conduct
* No Library Update Requests
* No Timeline Requests
* Please read through the documentation and FAQs
* Please be respectful of my time. I put a lot of effort into my libraries.  I try to support the issue requests within a day.  I'm not psychic, so please take the time to fill out the above form as complete as possible

diescalo, 2018-07-23:
Thank you for your fast answer! I will start taking a look to reactive programming (and re-think a bit the application too =) )

Thanks again.

closed 2018-07-23 by diescalo

MKGNZ, 2018-07-23:
It also seems that you already know the Characteristics that you need.
So you don't have to Discover them.
You can call 
```
Device.WhenConnected().Subscribe(async (connectedDevice) =>
{
    var _Service  = await connectedDevice.GetKnownService(Your service UUID here);
    txCharacteristic = await _Service.GetKnownCharacteristics(txCharID);
    rxCharacteristic = await _Service.GetKnownCharacteristics(rxCharID);
});

```

diescalo, 2018-07-24:
Hey, @MKGNZ ! I already tried things like that, however none of the methods can be awaited and they don't return the service or characteristic (which is what I was expecting too). This is a different way to do things that I should start get used to. Anyway, thank you very much!!

aritchie, 2018-07-24:
You can await observables (there is a set of extensions in the RX namespace).  You can also use things like .Take(1) and .ToTask().