aritchie/bluetoothle #219: Doesn't notify in case of failure operation fo iOS

vmirgorod, 2018-07-30:
Affected platforms
 - [Affects] iOS
 - [Maybe] Android
 - [Maybe] tvOS
 - [Maybe] macOS
 - [Maybe] UWP

## Version of OS(s) listed above with issue
All iOS

## Version of Library
6.1.3

## Expected Behaviour
Plugin should inform for a) Failures b) Completion of discovering on DiscoveringServices operation.

## Actual Behavior
Plugin informing only in case of discovered services, skipping all error cases.

## Notes

Probably discovering code should look like.
```
public override IObservable<IGattService> DiscoverServices() => Observable.Create<IGattService>(ob =>
        {
            Log.Info(BleLogCategory.Device, "service discovery hooked for device " + this.Uuid);
            var services = new Dictionary<Guid, IGattService>();

            var handler = new EventHandler<NSErrorEventArgs>((sender, args) =>
            {
                if (args.Error != null)
                    ob.OnError(new Exception(args.Error.LocalizedDescription));

                foreach (var native in this.peripheral.Services)
                {
                    var service = new GattService(this, native);
                    if (!services.ContainsKey(service.Uuid))
                    {
                        services.Add(service.Uuid, service);
                        ob.OnNext(service);
                    }
                }

               ob.Respond(null); // TODO: how to notify that discovering was ended?
            });
            this.peripheral.DiscoveredService += handler;
            this.peripheral.DiscoverServices();

            return () => this.peripheral.DiscoveredService -= handler;
        });
```

aritchie, 2018-07-30:
Respond does an OnCompleted.  I do agree with the error check but in all my time doing this, I have not seen an error occur at this level (it would currently manifest itself as a discovery timeout).

Are you experiencing a bug that you can comment further on?

Please refrain from blanket statements in the future (maybe on all platforms).  Stick to facts that can be worked with.

vmirgorod, 2018-07-30:
> Please refrain from blanket statements in the future (maybe on all platforms). Stick to facts that can be worked with.

Ok sorry, that why i marked them "maybe".

> Are you experiencing a bug that you can comment further on?

No i didn't. But probably it would be better to handle this error now (and not in some future release of iOS when Apple adds it to API).

> Respond does an OnCompleted.

Yes, but it seems that `OnCompleted` is never called. I didn't find it in code. Also in my experimental sample i'm doing:

```
       using PluginService = Plugin.BluetoothLE.IGattService;

        public Task DiscoverServicesAsync() {
            var taskCompletionSource = new TaskCompletionSource<bool>();

            Adapter.DiscoverServices().Subscribe(new AnonymousObserver<PluginService>(
                service => {
                    Console.WriteLine("Discovered " + service);
                },
                exception => taskCompletionSource.SetException(exception),
                () => {
                    taskCompletionSource.SetResult(true);
                    Console.WriteLine("Finished discovering");
                }));

            return taskCompletionSource.Task;
        }
```

Line "Finished discovering" is never called.

commit 91a94e2f (2018-08-01): Fix #219 - [ios] complete service discovery and check for errors
closed 2018-08-01 by aritchie