aritchie/bluetoothle #220: Start up Plugin.BluetootheLE project

Melitaa, 2018-08-07:
Hi,

I could not successfully built the samples.Droid program (I just open and built the project). I got the following error
[image] If anybody knows what is the reason for this error I I will be grateful for help.

Best regards
Melita

aritchie, 2018-08-08:
It compiles and deploys fine for me out of master.  Take a look at the build output and see if you are getting some sort of weird error.

You should also make sure you have the latest android sdk stuff.

aritchie, 2018-08-08:
Also, if you're deploying to an emulator like shown in your screenshot, you won't get very far since it doesn't have emulator bluetooth

aritchie, 2018-08-10:
No answer - closing

closed 2018-08-10 by aritchie