aritchie/bluetoothle #221: Notify stops after a couple of minutes

bluekuen, 2018-08-09:
## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [x] iOS

## Version of OS(s) listed above with issue
iOS 11.4.1 

## Version of Library
newest stable

## Expected Behaviour
Notifications don't stop

## Actual Behavior
when subscribing to Notifications with RegisterAndNotify on a notify Characterisitc,
everything works fine and i get result data but after an undeterministic amount of time
(sometimes it's after a minute, sometimes after 5 minutes ) the Notifications
stop and i get nothing anymore. I made sure that i'm connected and that my
notify Char is still notifying and not disposed, also the write notification still fires.

aritchie, 2018-08-09:
Please supply some sort of details!!!

1) code?
2) how often is your characteristic broadcasting in ms?
3) If every 3-400ms, how do you manage the backpressure?
4) anything in the output?
5) are you trapping your own errors in Subscribe?

bluekuen, 2018-08-09:
Thanks for answering!
1. I modified my code quite often the last couple hours. This is how i register and notify atm:
```
public void RegisterAndNotify()
        {
            logger.Debug("Register And Notify  started");
            try
            {

                _notifyChar = _characteristics.Where(x => x.Uuid == CHARACTERISTIC_NOTIFY_ID)
                    .FirstOrDefault();

                if (_notifyChar == null)
                {
                    logger.Error("NotifyChar is NULL");
                }
            }
            catch (Exception ex)
            {
                logger.Debug("Register And Notify  failed with Exception:" + ex);
                //throw;
            }
            finally
            {
                _notifySub = _notifyChar.EnableNotifications()
                    .Select(y => y.Characteristic.WhenNotificationReceived())
                    .Switch()
                    .Synchronize()
                    .Finally(() => { logger.Debug("notifysub  finallay"); })
                    .Subscribe(result =>
                    {
                        logger.Debug("Received Result Notification from notifySub");
                        foreach (var item in result.Data)
                        {
                            ReceivedData.Enqueue(item);
                        }
                    }
            , onError =>
            {
                logger.Debug("Error Occured on Notify Sub: Error: " + onError.StackTrace);
            },
            () =>
            {
                logger.Debug("OnCompleted");
            });
            }
        }
```
I send a byte array every 100 ms and add the result from a request to a Queu. Then i pick the result byte after byte until i have my answer. This is a cyclic progress in a native background thread. The progress
works synchronous:
I send a "request" byte array and the device is answering instantly. If i don't get an answer after 3 seconds, i throw a TimeOut Exception and send the same request again. This works for a couple of minutes but then i don't get any notifications anymore (no logger output either). 
In the notifyChar ouput Value stands "Timed out" while debugging.

aritchie, 2018-08-09:
You should try/catch in your Subscribe.  Is your queue thread safe?  Is there another thread reading from it as you’re pushing to it?

Does the sample project have this issue for your device?

bluekuen, 2018-08-09:
good point, i've added try catch to it. Yes it's a .net ConcurrentQueue and i'm reading from it after getting result data. 

I didn't test the sample project on my Device but that's a good idea, i will test it.

bluekuen, 2018-08-10:
@MKGNZ yes it's just test code and i honestly don't know why i put it in there anymore. the subscription hab been in try after checking if notifychar is null maybe i just wanted to get shure that subscription gets executed. But thank you nevertheless!!

I also think i found the issue. When i increase the time of my cyclic/polling funtion to 200 ms instead of 100 ms, the progress seems to work as expected. I guess the notifychar just coldn't keep up with notifying and stopped working. Because the Device and his Gatts are propietary and customized.
i will have a look at it and test it through. I also plan on integrating the sample projects.

Thank you very much for your help, i really appreciate it !

closed 2018-08-10 by bluekuen

aritchie, 2018-08-10:
It won’t be the notify characteristic that can’t handle it.  It will keep pumping data whether you are ready to process or not this creating backpressure.  In this case, your queue is a shared resource where the subscription and processing code are competing for the lock on it.

You should use an RX buffer and remove the synchronize, then process the buffer in another async process