aritchie/bluetoothle #222: Cant install lib in android platform v8.1

minaairsupport, 2018-08-10:
## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [] Android

## Version of OS(s) listed above with issue

## Version of Library
v6.13

## Expected Behaviour
Install normally 

## Actual Behavior
_If applicable, please include entire exception - type, message, stacktrace_
Attempting to gather dependency information for package 'Plugin.BluetoothLE.6.1.3' with respect to project '', targeting 'MonoAndroid,Version=v8.1'

aritchie, 2018-08-10:
Please supply the entire output of that message.  It works fine for me.

minaairsupport, 2018-08-10:
here is screenshot of issue 
[image]

aritchie, 2018-08-10:
That's not a nuget install message that you originally filed.  That is a post-compile issue of some sort.

minaairsupport, 2018-08-10:
may be this message help Circular dependency detected 'Acr.Core 2.1.0 => System.Reactive 3.1.1 => System.Reactive.PlatformServices 4.0.0 => System.Reactive 3.1.1'.

aritchie, 2018-08-10:
Yes - that is the problem which unfortunately is something to do with reactive extensions.  If you install System.Reactive 3.1.1 first, it should resolve the nuget issue

minaairsupport, 2018-08-11:
I tried to install System.Reactive but its dosent work

aritchie, 2018-08-11:
You need to supply some sort of details other than "it doesn't work" if you want help.  I'm not psychic and it works for me.

minaairsupport, 2018-08-11:
sorry I can install now the lib into android project but I face new issue , is this related to your lib 

Error: Exception while loading assemblies: System.IO.FileNotFoundException: Could not load assembly 'System.Threading.Tasks.Extensions, Version=4.1.1.0, Culture=neutral, PublicKeyToken=cc7b13ffcd2ddd51'. Perhaps it doesn't exist in the Mono for Android profile?
File name: 'System.Threading.Tasks.Extensions.dll'
  at Java.Interop.Tools.Cecil.DirectoryAssemblyResolver.Resolve (Mono.Cecil.AssemblyNameReference reference, Mono.Cecil.ReaderParameters parameters) [0x0009a] in /Users/builder/data/lanes/5945/dffc5912/source/monodroid/external/xamarin-android/external/Java.Interop/src/Java.Interop.Tools.Cecil/Java.Interop.Tools.Cecil/DirectoryAssemblyResolver.cs:229 
  at Java.Interop.Tools.Cecil.DirectoryAssemblyResolver.Resolve (Mono.Cecil.AssemblyNameReference reference) [0x00000] in /Users/builder/data/lanes/5945/dffc5912/source/monodroid/external/xamarin-android/external/Java.Interop/src/Java.Interop.Tools.Cecil/Java.Interop.Tools.Cecil/DirectoryAssemblyResolver.cs:179 
  at Xamarin.Android.Tasks.ResolveAssemblies.AddAssemblyReferences (Java.Interop.Tools.Cecil.DirectoryAssemblyResolver resolver, System.Collections.Generic.ICollection`1[T] assemblies, Mono.Cecil.AssemblyDefinition assembly, System.Boolean topLevel) [0x0014a] in <96027da06dd5411688a31011072e256f>:0 
  at Xamarin.Android.Tasks.ResolveAssemblies.Execute (Java.Interop.Tools.Cecil.DirectoryAssemblyResolver resolver) [0x00237] in <96027da06dd5411688a31011072e256f>:0  (Autotell.Android)

aritchie, 2018-08-11:
Nope

closed 2018-08-11 by aritchie