aritchie/bluetoothle #223: GetKnownService

samosim, 2018-08-12:
Hi everyone,

I have one question about GetKnownService method which can be properly executed when BLE device is connected.  I would like to implement this method in the program when the device is connected.  I know that I can call the GetKnownService in the following way:

```
this.device
                .WhenStatusChanged()
                .ObserveOn(RxApp.MainThreadScheduler)
                .Subscribe(status =>
                {
                    switch (status)
                    {
                        case ConnectionStatus.Connecting:
                            this.ConnectText = "Connecting";
                            break;

                        case ConnectionStatus.Connected:
                            this.ConnectText = "Connected";
                            this.WaitFlag     = true;
                             this.device.GetKnownService(App.MassService).Subscribe(ser => MassService = ser.Uuid.ToString());                            
                            break;

                        case ConnectionStatus.Disconnected:
                            this.ConnectText = "Disconnected";
                            try
                            {
                                //this.GattCharacteristics.Clear();
                            }
                            catch (Exception ex)
                            {
                                Console.WriteLine(ex);
                            }

                            break;
                    }
                }
                );
```

But I would like to know how I can call this method directly in Contructor() without nest implementation. 

Best regards

Samo

samosim, 2018-08-12:
I guess that I found out a solution which is the following code:

this.device.WhenConnected().Subscribe(dv => dv.GetKnownService(App.MassService).Subscribe(ser => MassService = ser.Uuid.ToString()));

I will grateful for any comments

Best regards

Samo

aritchie, 2018-08-12:
You should look at the extension methods built off the device.  They focus more on the characteristics but that is really what you need.

You really need to be careful about storing references to services and characteristics as they are bad after the device disconnects for any reason

samosim, 2018-08-13:
Thanks for help Aritchie.

closed 2018-08-13 by aritchie