aritchie/bluetoothle #224: On Android, after service change, it is impossible to connect / discover services / characteristics

gunpuz, 2018-08-13:
## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [] iOS
 - [*] Android
 - [] tvOS
 - [] macOS
 - [] .NET Standard
 - UWP IS NOT SUPPORTED DURING THE BETA PERIOD - ONLY PRs ARE WELCOME FOR NOW

## Version of OS(s) listed above with issue

## Version of Library

## Expected Behaviour

Works like the first and all of other times. Device should normally connect.

## Actual Behavior
_If applicable, please include entire exception - type, message, stacktrace_

After BLE device restart and services changed, it is impossible to connect, or discover services, just my timeout is hit, nothing really happens.

It could be related to 
[https://stackoverflow.com/questions/22596951/how-to-programmatically-force-bluetooth-low-energy-service-discovery-on-android](https://stackoverflow.com/questions/22596951/how-to-programmatically-force-bluetooth-low-energy-service-discovery-on-android)

## Code of Conduct
* No Library Update Requests
* No Timeline Requests
* Please read through the documentation and FAQs
* Please be respectful of my time. I put a lot of effort into my libraries.  I try to support the issue requests within a day.  I'm not psychic, so please take the time to fill out the above form as complete as possible

aritchie, 2018-08-13:
Please include sample code and an android device+OS so I have something to work with!

Also, you say it isn’t connecting at all or just the discovery portion is timing out

gunpuz, 2018-08-13:
Unfortunately, the full code of what i am writing is not opensourced just yet.
* Huawei, Android 8.0.0, EMUI 8.0.0, Build: FIG-LX1 8.0.0.148
What i am actually writing: https://github.com/NordicSemiconductor/Android-DFU-Library

Here is what is happening: 
Connect to device, enable indications, write some.
Wait until device disconnects, cancel connection
--- MEANWHILE device restarts with MAC address + 1, services changed
Connect to new device, listen notifications, write some stuff
wait disconnect, cancel connection
-- MEANWHILE device restarts with MAC address and previous services

At second an all other attempts (at some point it starts working), it fails, i have 10 seconds timeout. However, when connecting from [https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp], it works just fine. And what is interesting, after connecting from NRF connect and disconnecting, my app starts to work, so i tend to think that NRF connect does something. And i think it could be https://github.com/NordicSemiconductor/Android-DFU-Library/blob/5cac2ce0e6f8fbf80f581862066ed2f56f58e3ec/dfu/src/main/java/no/nordicsemi/android/dfu/DfuBaseService.java#L1414

I happened to modify clone and modify your library, it seems, that it does not work 100%.

        
        public void RefreshGatt()
        {
            // https://stackoverflow.com/questions/22596951/how-to-programmatically-force-bluetooth-low-energy-service-discovery-on-android
            try
            {
                Log.Warn(BleLogCategory.Device, "Try to clear Android cache");
                Method refreshMethod = this.Gatt.Class.GetMethod("refresh", new Class[0]);
                if (refreshMethod != null)
                {
                    bool result = (bool)refreshMethod.Invoke(this.Gatt, new Java.Lang.Object[0]);
                    Log.Warn(BleLogCategory.Device, "Cache result = " + result);
                }
            }
            catch (Exception ex)
            {

            }
        }

What i do noticed, that NRF connect says that indications on one specific characteristic (I enable them, but as soon as not needed i call disableNotifications()) are enabled after my procedure, i will investigate it. Maybe it somehow blocks the cache cleaning?

gunpuz, 2018-08-13:
About the example code, you would actually need some device who changes services i suppose, if you are trying to reproduce.

closed 2018-08-18 by aritchie