aritchie/bluetoothle #225: Unpair Device

Paolo34, 2018-08-13:
There is a method that implements UnpairingRequest?
Because the connection with devices requiring the pairing works only if I press CONNECT button and then I press PAIR DEVICE button(if I press PAIR DEVICE button and then I press CONNECT button devices don't connect each other). When devices are already paired the connection doesn't work, so every time I disconnect a device I would like to unpair it.
Thanks

aritchie, 2018-08-13:
There is no "unpair" function on any platform that this library supports

closed 2018-08-13 by aritchie

Paolo34, 2018-08-13:
It is normal that paired devices cannot be connected?

aritchie, 2018-08-13:
Nope.  I have a paired BLE device that works fine.  It isn’t really common for BLE device require any sort of pairing though

Paolo34, 2018-08-13:
I don't know why, but can establish connection only if after ask for a connection ask also for a pairing, even if the pairing is not required by the other device

aritchie, 2018-08-13:
I don't know either because you didn't fill in any of the issue template and I have no code to even make a guess :)

Paolo34, 2018-08-13:
I'm using your code without any change and i have deployed it in an Android 7.0 device

aritchie, 2018-08-13:
Excellent, no I can try to help.  

Does your host device have GATT setup on it?

aritchie, 2018-08-13:
Is there any errors or output relevant in the output window?
Can you give me any details about the host device?

Paolo34, 2018-08-13:
When i press CONNECT the output window is:
08-13 20:40:02.012 D/ViewRootImpl@21766f5[MainActivity](17238): ViewPostImeInputStage processPointer 0
08-13 20:40:02.126 D/ViewRootImpl@21766f5[MainActivity](17238): ViewPostImeInputStage processPointer 1
08-13 20:40:02.134 D/BluetoothManager(17238): getConnectionState()
08-13 20:40:02.135 D/BluetoothManager(17238): getConnectedDevices
08-13 20:40:02.175 D/BluetoothGatt(17238): connect() - device: 50:B1:94:B7:1E:XX, auto: true
08-13 20:40:02.175 D/BluetoothAdapter(17238): isSecureModeEnabled
08-13 20:40:02.176 D/BluetoothGatt(17238): registerApp()
08-13 20:40:02.176 D/BluetoothGatt(17238): registerApp() - UUID=ba90a2b7-c022-41b8-a87c-6cc71129d957
08-13 20:40:02.281 D/BluetoothGatt(17238): onClientRegistered() - status=0 clientIf=5
08-13 20:40:04.588 D/ViewRootImpl@21766f5[MainActivity](17238): ViewPostImeInputStage processPointer 0
08-13 20:40:04.727 D/ViewRootImpl@21766f5[MainActivity](17238): ViewPostImeInputStage processPointer 1
08-13 20:40:04.775 I/BluetoothDevice(17238): createBond() for device 50:B1:94:B7:1E:XX called by pid: 17238 tid: 17238
08-13 20:40:05.259 D/BluetoothGatt(17238): onClientConnectionState() - status=0 clientIf=5 device=50:B1:94:B7:1E:XX
and the device doesn't connect to my app. Then i press PAIR DEVICE and the output is:
Thread started:  #7
[0:] 
08-13 20:40:05.468 D/Mono    (17238): DllImport attempting to load: '__Internal'.
08-13 20:40:05.468 D/Mono    (17238): DllImport loaded library '(null)'.
08-13 20:40:05.468 D/Mono    (17238): DllImport searching in: '__Internal' ('(null)').
08-13 20:40:05.468 D/Mono    (17238): Searching for 'monodroid_get_system_property'.
08-13 20:40:05.468 D/Mono    (17238): Probing 'monodroid_get_system_property'.
08-13 20:40:05.468 D/Mono    (17238): Found as 'monodroid_get_system_property'.
08-13 20:40:05.468 D/Mono    (17238): DllImport searching in: '__Internal' ('(null)').
08-13 20:40:05.468 D/Mono    (17238): Searching for 'monodroid_free'.
08-13 20:40:05.468 D/Mono    (17238): Probing 'monodroid_free'.
08-13 20:40:05.468 D/Mono    (17238): Found as 'monodroid_free'.
[0:] 
[0:] 
08-13 20:40:05.608 D/BluetoothGatt(17238): discoverServices() - device: 50:B1:94:B7:1E:XX
08-13 20:40:05.866 D/BluetoothGatt(17238): onClientConnParamsChanged() - Device=50:B1:94:B7:1E:XX interval=6 status=0
08-13 20:40:05.957 D/Mono    (17238): Assembly Ref addref Acr.UserDialogs[0xcb92b3c0] -> Xamarin.Android.Support.Design[0xcb92bf00]: 3
08-13 20:40:06.013 D/TextView(17238): setTypeface with style : 0
08-13 20:40:07.019 D/BluetoothGatt(17238): onSearchComplete() = Device=50:B1:94:B7:1E:XX Status=0
08-13 20:40:07.305 I/art     (17238): Starting a blocking GC Explicit
08-13 20:40:07.328 I/art     (17238): Explicit concurrent mark sweep GC freed 5623(306KB) AllocSpace objects, 0(0B) LOS objects, 40% free, 10MB/17MB, paused 242us total 23.161ms
08-13 20:40:07.329 D/Mono    (17238): GC_TAR_BRIDGE bridges 183 objects 6593 opaque 2634 colors 183 colors-bridged 183 colors-visible 183 xref 3 cache-hit 0 cache-semihit 0 cache-miss 0 setup 0.04ms tarjan 2.71ms scc-setup 0.13ms gather-xref 0.01ms xref-setup 0.00ms cleanup 0.15ms
08-13 20:40:07.329 D/Mono    (17238): GC_BRIDGE: Complete, was running for 25.37ms
08-13 20:40:07.329 D/Mono    (17238): GC_MINOR: (Nursery full) time 15.60ms, stw 16.83ms promoted 1054K major size: 2064K in use: 1337K los size: 1024K in use: 191K
08-13 20:40:07.473 D/AbsListView(17238): viewType is heaer or footer
and finally the connection is establish.

Paolo34, 2018-08-13:
in that case the host was an iphone but the same thing happens also with apple watches, my smart scales (Beurer BF700) and an STM32 Nucleo with a BLE expansion Board (that is my really purpose for which I'm developing an app)

aritchie, 2018-08-13:
The sample app isn’t fully setup for end to end pairing at the moment.  It something I will look to implement in the future.  The APIs are all built in and I was able to work the device fine if the device was paired beforehand.