aritchie/bluetoothle #227: Can't read RSSI value

jeremym33, 2018-08-27:
## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [] iOS
 - [X] Android
 - [] tvOS
 - [] macOS
 - [] .NET Standard
 - UWP IS NOT SUPPORTED DURING THE BETA PERIOD - ONLY PRs ARE WELCOME FOR NOW

## Version of OS(s) listed above with issue
Version 6.0 +

## Version of Library
6.1.3

## Expected Behaviour

if (_connectedDevice?.Device.Status == ConnectionStatus.Connected)
            {
                _connectedDevice.Device.WhenReadRssiContinuously().Subscribe(value =>
                {
                     Rssi = value.ToString();
                });
            }

The device is Connected but the Rssi value is never set.
I can perform write et read Ble Request with the device.
In version 5.2.0 of library i can read Rssi.

Thank you for your works.

## Actual Behavior
_If applicable, please include entire exception - type, message, stacktrace_

## Code of Conduct
* No Library Update Requests
* No Timeline Requests
* Please read through the documentation and FAQs
* Please be respectful of my time. I put a lot of effort into my libraries.  I try to support the issue requests within a day.  I'm not psychic, so please take the time to fill out the above form as complete as possible

aritchie, 2018-08-28:
Working fine for me.  Please provide a failing test sample with zero abstractions around it and I can take a look.

You don't need to check device connection with WhenReadRssiContinuously as it will fire each time the rehook

jeremym33, 2018-08-28:
After rebuild the app. It's work fine. Sorry for the inconvenience.

closed 2018-08-28 by jeremym33