aritchie/bluetoothle #230: Package not detecting Manifest permissions

Rousseau38, 2018-08-29:
## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [] iOS
 - [X] Android
 - [] tvOS
 - [] macOS
 - [X] .NET Standard
 - UWP IS NOT SUPPORTED DURING THE BETA PERIOD - ONLY PRs ARE WELCOME FOR NOW

## Version of OS(s) listed above with issue
Android version 8.0

## Version of Library
6.2.0

## Expected Behaviour
After calling:

```
                var scanner = CrossBleAdapter.Current.Scan().Subscribe(result =>
                {

                });
```

app should start scanning and detect devices. 

## Actual Behavior
project fails with this error in output (in Bold): 

08-29 13:27:16.421 D/OpenGLRenderer(18475): HWUI GL Pipeline
08-29 13:27:16.446 D/Mono    (18475): Assembly Ref addref ReactiveBLE[0x79a5091e00] -> Plugin.BluetoothLE[0x79a5091d00]: 2
Resolved pending breakpoint at 'ReactBLE.cs:16,1' to void ReactiveBLE.ReactBLE.<>c.<ScanForDevices>b__0_0 (Plugin.BluetoothLE.IScanResult result) [0x00001].
08-29 13:27:16.486 D/Mono    (18475): Assembly Ref addref ReactiveBLE[0x79a5091e00] -> System.Reactive[0x79a5091f00]: 2
Resolved pending breakpoint at 'ReactBLE.cs:14,1' to void ReactiveBLE.ReactBLE.ScanForDevices () [0x00002].
Resolved pending breakpoint at 'ReactBLE.cs:20,1' to void ReactiveBLE.ReactBLE.ScanForDevices () [0x00036].
08-29 13:31:39.431 D/Mono    (18475): Assembly Ref addref Plugin.BluetoothLE[0x79a5091d00] -> System.Reactive[0x79a5091f00]: 3
08-29 13:31:39.439 D/Mono    (18475): Assembly Ref addref Plugin.BluetoothLE[0x79a5091d00] -> System.Core[0x79a4f32780]: 7
08-29 13:31:39.455 D/BluetoothManager(18475): getDevicesMatchingConnectionStates
08-29 13:31:39.476 D/Mono    (18475): DllImport searching in: '__Internal' ('(null)').
08-29 13:31:39.476 D/Mono    (18475): Searching for 'java_interop_jnienv_call_boolean_method'.
08-29 13:31:39.476 D/Mono    (18475): Probing 'java_interop_jnienv_call_boolean_method'.
08-29 13:31:39.477 D/Mono    (18475): Found as 'java_interop_jnienv_call_boolean_method'.
Thread started:  #3
08-29 13:31:39.654 D/BluetoothAdapter(18475): isLeEnabled(): ON
08-29 13:31:39.659 D/BluetoothLeScanner(18475): onScannerRegistered() - status=0 scannerId=12 mScannerId=0
08-29 13:31:39.665 W/Binder  (18475): Caught a RuntimeException from the binder stub implementation.
**08-29 13:31:39.665 W/Binder  (18475): java.lang.SecurityException: Need ACCESS_COARSE_LOCATION or ACCESS_FINE_LOCATION permission to get scan results**
08-29 13:31:39.665 W/Binder  (18475): 	at android.os.Parcel.readException(Parcel.java:1942)
08-29 13:31:39.665 W/Binder  (18475): 	at android.os.Parcel.readException(Parcel.java:1888)
08-29 13:31:39.665 W/Binder  (18475): 	at android.bluetooth.IBluetoothGatt$Stub$Proxy.startScan(IBluetoothGatt.java:918)
08-29 13:31:39.665 W/Binder  (18475): 	at android.bluetooth.le.BluetoothLeScanner$BleScanCallbackWrapper.onScannerRegistered(BluetoothLeScanner.java:431)
08-29 13:31:39.665 W/Binder  (18475): 	at android.bluetooth.le.IScannerCallback$Stub.onTransact(IScannerCallback.java:56)
08-29 13:31:39.665 W/Binder  (18475): 	at android.os.Binder.execTransact(Binder.java:674)
08-29 13:31:41.715 I/Adreno  (18475): QUALCOMM build                   : 7142022, Ib5823dd10c
08-29 13:31:41.715 I/Adreno  (18475): Build Date                       : 06/23/17
08-29 13:31:41.715 I/Adreno  (18475): OpenGL ES Shader Compiler Version: EV031.18.00.00
08-29 13:31:41.715 I/Adreno  (18475): Local Branch                     : O11A
08-29 13:31:41.715 I/Adreno  (18475): Remote Branch                    : 
08-29 13:31:41.715 I/Adreno  (18475): Remote Branch                    : 
08-29 13:31:41.715 I/Adreno  (18475): Reconstruct Branch               : 
08-29 13:31:41.715 I/vndksupport(18475): sphal namespace is not configured for this process. Loading /vendor/lib64/hw/gralloc.msm8994.so from the current namespace instead.
08-29 13:31:41.741 I/Adreno  (18475): PFP: 0x00000000, ME: 0x00000000
08-29 13:31:41.758 I/OpenGLRenderer(18475): Initialized EGL, version 1.4
08-29 13:31:41.758 D/OpenGLRenderer(18475): Swap behavior 2
08-29 13:31:41.808 D/Mono    (18475): DllImport searching in: '__Internal' ('(null)').
08-29 13:31:41.808 D/Mono    (18475): Searching for 'java_interop_jnienv_call_nonvirtual_float_method_a'.
08-29 13:31:41.808 D/Mono    (18475): Probing 'java_interop_jnienv_call_nonvirtual_float_method_a'.
08-29 13:31:41.808 D/Mono    (18475): Found as 'java_interop_jnienv_call_nonvirtual_float_method_a'.
08-29 13:31:41.836 I/vndksupport(18475): sphal namespace is not configured for this process. Loading /vendor/lib64/hw/gralloc.msm8994.so from the current namespace instead.

Here is my Manifest: 
```
<?xml version="1.0" encoding="utf-8"?>
<manifest xmlns:android="http://schemas.android.com/apk/res/android" android:versionCode="1" android:versionName="1.0" package="com.companyname.ReactiveBLE" android:installLocation="auto">
	<uses-sdk android:minSdkVersion="19" android:targetSdkVersion="26" />
	<uses-permission android:name="android.permission.BLUETOOTH" />
	<uses-permission android:name="android.permission.BLUETOOTH_ADMIN" />
	<uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION" />
	<uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" />
	<application android:label="ReactiveBLE.Android"></application>
</manifest>
```

I h

I am following the usage guide on this page:
https://github.com/aritchie/bluetoothle

This is with a new project with Xamarin.Forms refrenced. 

Thank you for your support.

aritchie, 2018-08-29:
You still need to request permission to use the location services which is external to this library.  You can use something like permission plugin

closed 2018-08-29 by Rousseau38

michaelkollmann, 2018-08-30:
I think I'm having the same issue here. Why would I need to request permission for location just to to get bluetooth to work?

In the "SETUP" guid it says "" regarding the permission.
I'm not interested in either the name or the address. Is there any way to do this without the user having to allow the location permission first?

Also, please make this more clear to understand in the SETUP guide. This should be part of the documentation imho

aritchie, 2018-08-30:
No - this is simply the way android permissions about Bluetooth work due privacy reasons.  

As for docs, if you don't like them, please send a PR.  I'm not trying to "sell" this library.  If it adds value to you, great, if it doesn't, great.  I do docs when I feel like it which is rarely since it isn't exactly fun for me.