aritchie/bluetoothle #231: Advertiser Start - not working

realex82, 2018-09-04:
Hello,
when i exceute the code for start server on my app it not working.
Code:

> 
          List<Guid> Lst_g = new List<Guid>();
           Lst_g.Add(Guid.Parse("A495FF20-C5B1-4B44-B512-1370F02D74DE"));
     
          await this.BuildServer();
            adapter.Advertiser.Start(new AdvertisementData
            {
                LocalName = "My GATT",
                ServiceUuids = Lst_g

            });
....
Error at adapter.Advertiser: 
**System.NullReferenceException: Object reference not set to an instance of an object**
I've make debug with Visual studio and adapter.Advertiser is null 

How can i risolve it ?

Thanks
Alex

realex82, 2018-09-06:
Hello, nothing? Only to know if I have to move on other things

Thanks
Alex

aritchie, 2018-09-06:
I tend ban people for bumps.  It has only been a couple of days.  You can be patient and wait for when I have time to look into or since it is OSS, you can look into it yourself.

realex82, 2018-09-06:
ok, but not to put a hurry, in fact you can put the time that you consider appropriate. I have intended in case I had not written well the error and did not understand what I had written

That's all
In fact I have also thanked you

rbrian, 2018-09-06:
Looks like there's nothing setting Advertiser in the Adapter ctor on iOS, like there is on other platforms. Will fix and test locally whilst waiting for an official fix. Let me know if you want a PR on it.

realex82, 2018-09-07:
Thanks for the reply. for me, any solution is fine, local or otherwise.

Alex

commit d2b07284 (2018-09-11): fix #231 - advertiser NRE
closed 2018-09-11 by aritchie