aritchie/bluetoothle #232: Write sometimes brakes things

gunpuz, 2018-09-06:
## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [] iOS
 - [*] Android
 - [] tvOS
 - [] macOS
 - [] .NET Standard
 - UWP IS NOT SUPPORTED DURING THE BETA PERIOD - ONLY PRs ARE WELCOME FOR NOW

## Version of OS(s) listed above with issue
Huawei Android 7.0, EMUI 5.0

## Version of Library

6.2.0
## Expected Behaviour

## Actual Behavior
_If applicable, please include entire exception - type, message, stacktrace_

Not sure from where this is comming, because no application level stack trace is shown. Well obviously it comes from WriteCharacteristic, but like why?
```
09-06 13:50:53.912 I/Choreographer(22749): Skipped 595 frames!  The application may be doing too much work on its main thread.
09-06 13:50:53.915 E/OpenGLRenderer(22749): allen debug liyu Key: 5242944
Unhandled Exception:

Java.Lang.NullPointerException: Attempt to invoke interface method 'java.util.Iterator java.lang.Iterable.iterator()' on a null object reference

Thread started: <Thread Pool> #13
Thread started: <Thread Pool> #14
Thread started: <Thread Pool> #15
Thread started: <Thread Pool> #16
Thread started: <Thread Pool> #17
Thread started: <Thread Pool> #18
Thread started: <Thread Pool> #19
09-06 13:51:02.790 I/MonoDroid(22749): UNHANDLED EXCEPTION:
09-06 13:51:02.807 I/MonoDroid(22749): Java.Lang.NullPointerException: Attempt to invoke interface method 'java.util.Iterator java.lang.Iterable.iterator()' on a null object reference
09-06 13:51:02.807 I/MonoDroid(22749):   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw () [0x0000c] in <f32579baafc1404fa37ba3ec1abdc0bd>:0 
09-06 13:51:02.807 I/MonoDroid(22749):   at Java.Interop.JniEnvironment+InstanceMethods.CallBooleanMethod (Java.Interop.JniObjectReference instance, Java.Interop.JniMethodInfo method, Java.Interop.JniArgumentValue* args) [0x00069] in <7802aa64ad574c33adca332a3fa9706a>:0 
09-06 13:51:02.807 I/MonoDroid(22749):   at Java.Interop.JniPeerMembers+JniInstanceMethods.InvokeAbstractBooleanMethod (System.String encodedMember, Java.Interop.IJavaPeerable self, Java.Interop.JniArgumentValue* parameters) [0x00014] in <7802aa64ad574c33adca332a3fa9706a>:0 
09-06 13:51:02.807 I/MonoDroid(22749):   at Android.Bluetooth.BluetoothGatt.WriteCharacteristic (Android.Bluetooth.BluetoothGattCharacteristic characteristic) [0x00031] in <263adecfa58f4c449f1ff56156d886fd>:0 
09-06 13:51:02.807 I/MonoDroid(22749):   at Plugin.BluetoothLE.GattCharacteristic+<>c__DisplayClass7_1.<Write>b__2 () [0x00037] in C:\dev\acr\bluetoothle\Plugin.BluetoothLE\Platforms\Android\GattCharacteristic.cs:91 
09-06 13:51:02.807 I/MonoDroid(22749):   at Acr.PlatformImpl+<>c__DisplayClass0_0.<InvokeOnMainThread>b__0 (System.Object _) [0x00000] in <0e46cff086394156ac1f511db60e9e0d>:0 
09-06 13:51:02.807 I/MonoDroid(22749):   at Android.App.SyncContext+<>c__DisplayClass2_0.<Post>b__0 () [0x00000] in <263adecfa58f4c449f1ff56156d886fd>:0 
09-06 13:51:02.807 I/MonoDroid(22749):   at Java.Lang.Thread+RunnableImplementor.Run () [0x00008] in <263adecfa58f4c449f1ff56156d886fd>:0 
09-06 13:51:02.807 I/MonoDroid(22749):   at Java.Lang.IRunnableInvoker.n_Run (System.IntPtr jnienv, System.IntPtr native__this) [0x00008] in <263adecfa58f4c449f1ff56156d886fd>:0 
09-06 13:51:02.807 I/MonoDroid(22749):   at (wrapper dynamic-method) System.Object.b8bc66b3-b4a7-49d8-8f43-e12473fef153(intptr,intptr)
09-06 13:51:02.807 I/MonoDroid(22749):   --- End of managed Java.Lang.NullPointerException stack trace ---
09-06 13:51:02.807 I/MonoDroid(22749): java.lang.NullPointerException: Attempt to invoke interface method 'java.util.Iterator java.lang.Iterable.iterator()' on a null object reference
09-06 13:51:02.807 I/MonoDroid(22749): 	at android.os.Parcel.readException(Parcel.java:1671)
09-06 13:51:02.807 I/MonoDroid(22749): 	at android.os.Parcel.readException(Parcel.java:1618)
09-06 13:51:02.807 I/MonoDroid(22749): 	at android.bluetooth.IBluetoothGatt$Stub$Proxy.writeCharacteristic(IBluetoothGatt.java:1042)
09-06 13:51:02.807 I/MonoDroid(22749): 	at android.bluetooth.BluetoothGatt.writeCharacteristic(BluetoothGatt.java:899)
09-06 13:51:02.807 I/MonoDroid(22749): 	at mono.java.lang.RunnableImplementor.n_run(Native Method)
09-06 13:51:02.807 I/MonoDroid(22749): 	at mono.java.lang.RunnableImplementor.run(RunnableImplementor.java:30)
09-06 13:51:02.807 I/MonoDroid(22749): 	at android.os.Handler.handleCallback(Handler.java:761)
09-06 13:51:02.807 I/MonoDroid(22749): 	at android.os.Handler.dispatchMessage(Handler.java:98)
09-06 13:51:02.807 I/MonoDroid(22749): 	at android.os.Looper.loop(Looper.java:156)
09-06 13:51:02.807 I/MonoDroid(22749): 	at android.app.ActivityThread.main(ActivityThread.java:6523)
09-06 13:51:02.807 I/MonoDroid(22749): 	at java.lang.reflect.Method.invoke(Native Method)
09-06 13:51:02.807 I/MonoDroid(22749): 	at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:941)
09-06 13:51:02.807 I/MonoDroid(22749): 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:831)
09-06 13:51:02.807 I/MonoDroid(22749): 
An unhandled exception occured.

Thread finished: <Thread Pool> #3
The thread 0x3 has exited with code 0 (0x0).
Thread finished: <Thread Pool> #15
The thread 0xf has exited with code 0 (0x0).
Thread finished: <Thread Pool> #18
[... 43 lines cut ...]
09-06 13:55:16.623 E/mono-rt (22749): 	at mono.java.lang.RunnableImplementor.n_run(Native Method)
09-06 13:55:16.623 E/mono-rt (22749): 	at mono.java.lang.RunnableImplementor.run(RunnableImplementor.java:30)
09-06 13:55:16.623 E/mono-rt (22749): 	at android.os.Handler.handleCallback(Handler.java:761)
09-06 13:55:16.623 E/mono-rt (22749): 	at android.os.Handler.dispatchMessage(Handler.java:98)
09-06 13:55:16.623 E/mono-rt (22749): 	at android.os.Looper.loop(Looper.java:156)
09-06 13:55:16.623 E/mono-rt (22749): 	at android.app.ActivityThread.main(ActivityThread.java:6523)
09-06 13:55:16.623 E/mono-rt (22749): 	at java.lang.reflect.Method.invoke(Native Method)
09-06 13:55:16.623 E/mono-rt (22749): 	at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:941)
09-06 13:55:16.623 E/mono-rt (22749): 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:831)
09-06 13:55:16.623 E/mono-rt (22749): 
Debugger Connection Lost: Debugger lost connection to the running application. Likely this means the application terminated unexpectedly.

More details can usually be found in the device's debug log:

http://developer.xamarin.com/guides/android/deployment,_testing,_and_metrics/android_debug_log/
Exception of type 'Mono.Debugger.Soft.VMDisconnectedException' was thrown.

```

## Code of Conduct
* No Library Update Requests
* No Timeline Requests
* Please read through the documentation and FAQs
* If you have a complaint about the docs, FIX IT - DO NOT FILE IT HERE!  I won't guarantee everything is always 100% up-to-date since I am just 1 person.
* Please be respectful of my time. I put a lot of effort into my libraries.  I try to support the issue requests within a day.  I'm not psychic, so please take the time to fill out the above form as complete as possible

aritchie, 2018-09-06:
The internals of the bluetooth stack are crashing which usually means a blip in the connection.  You should always have an exception trap on read/write ops

gunpuz, 2018-09-06:
I think that i may have all the [try,caches]. It happened only once. Never seen it before and after. I am wondering why there are no stacktraces from my application?
Only stack from C# is this line: `at Plugin.BluetoothLE.GattCharacteristic+<>c__DisplayClass7_1.<Write>b__2 () [0x00037] in C:\dev\acr\bluetoothle\Plugin.BluetoothLE\Platforms\Android\GattCharacteristic.cs:91`

aritchie, 2018-09-06:
The code is executed in a background thread.   You won’t see your app code.

gunpuz, 2018-09-07:
So i checked last night, that i used ble in event handler which is async and did not catch that it is called quite a lot, well i fixed it. I guess that i was not handling sync BLE use

To sum up : i probably would call it a library misuse. Keep up the good work!

I saw that you catch exceptions on connect(); [connect](https://github.com/aritchie/bluetoothle/blob/ed5ba23a9cf091c6687ac2e5fa362e6fc7c7b17e/Plugin.BluetoothLE/Platforms/Android/Internals/DeviceContext.cs#L43) Although, not sure if it is safe. But i realized, i am not catching anything on cancel connection

What actually happened:

eventhandler =>
   ```
            device.Connect();
            try
            {
                await device.ConnectWait().Timeout(timeout);
                await device.WriteCharacteristic(service, char, command).Timeout(timeout).Retry(2);

            }
            catch(Exception ex)
            {
                Debug.WriteLine(ex);
            }
            device.CancelConnection();
```

gunpuz, 2018-09-10:
Update:
not sure from what "my code" this is coming from, but seems like Read operation has something similar stacktrace.
Still not sure what i am doing wrong :(
```
Unhandled Exception:

Java.Lang.NullPointerException: Attempt to invoke interface method 'java.util.Iterator java.lang.Iterable.iterator()' on a null object reference

Thread started: <Thread Pool> #401
Thread started: <Thread Pool> #402
Thread started: <Thread Pool> #403
Thread started: <Thread Pool> #404
Thread started: <Thread Pool> #405
09-10 18:31:07.175 I/MonoDroid( 6822): UNHANDLED EXCEPTION:
09-10 18:31:07.180 I/MonoDroid( 6822): Java.Lang.NullPointerException: Attempt to invoke interface method 'java.util.Iterator java.lang.Iterable.iterator()' on a null object reference
09-10 18:31:07.180 I/MonoDroid( 6822):   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw () [0x0000c] in <f32579baafc1404fa37ba3ec1abdc0bd>:0 
09-10 18:31:07.180 I/MonoDroid( 6822):   at Java.Interop.JniEnvironment+InstanceMethods.CallBooleanMethod (Java.Interop.JniObjectReference instance, Java.Interop.JniMethodInfo method, Java.Interop.JniArgumentValue* args) [0x00069] in <7802aa64ad574c33adca332a3fa9706a>:0 
09-10 18:31:07.180 I/MonoDroid( 6822):   at Java.Interop.JniPeerMembers+JniInstanceMethods.InvokeAbstractBooleanMethod (System.String encodedMember, Java.Interop.IJavaPeerable self, Java.Interop.JniArgumentValue* parameters) [0x00014] in <7802aa64ad574c33adca332a3fa9706a>:0 
09-10 18:31:07.180 I/MonoDroid( 6822):   at Android.Bluetooth.BluetoothGatt.ReadCharacteristic (Android.Bluetooth.BluetoothGattCharacteristic characteristic) [0x00031] in <263adecfa58f4c449f1ff56156d886fd>:0 
09-10 18:31:07.180 I/MonoDroid( 6822):   at Plugin.BluetoothLE.GattCharacteristic+<>c__DisplayClass8_0.<Read>b__2 () [0x00000] in C:\dev\acr\bluetoothle\Plugin.BluetoothLE\Platforms\Android\GattCharacteristic.cs:117 
09-10 18:31:07.180 I/MonoDroid( 6822):   at Acr.PlatformImpl+<>c__DisplayClass0_0.<InvokeOnMainThread>b__0 (System.Object _) [0x00000] in <0e46cff086394156ac1f511db60e9e0d>:0 
09-10 18:31:07.180 I/MonoDroid( 6822):   at Android.App.SyncContext+<>c__DisplayClass2_0.<Post>b__0 () [0x00000] in <263adecfa58f4c449f1ff56156d886fd>:0 
09-10 18:31:07.180 I/MonoDroid( 6822):   at Java.Lang.Thread+RunnableImplementor.Run () [0x00008] in <263adecfa58f4c449f1ff56156d886fd>:0 
09-10 18:31:07.180 I/MonoDroid( 6822):   at Java.Lang.IRunnableInvoker.n_Run (System.IntPtr jnienv, System.IntPtr native__this) [0x00008] in <263adecfa58f4c449f1ff56156d886fd>:0 
09-10 18:31:07.180 I/MonoDroid( 6822):   at (wrapper dynamic-method) System.Object.f36023d6-e2a3-4dab-bac1-a422ff3fd481(intptr,intptr)
09-10 18:31:07.180 I/MonoDroid( 6822):   --- End of managed Java.Lang.NullPointerException stack trace ---
09-10 18:31:07.180 I/MonoDroid( 6822): java.lang.NullPointerException: Attempt to invoke interface method 'java.util.Iterator java.lang.Iterable.iterator()' on a null object reference
09-10 18:31:07.180 I/MonoDroid( 6822): 	at android.os.Parcel.readException(Parcel.java:1671)
09-10 18:31:07.180 I/MonoDroid( 6822): 	at android.os.Parcel.readException(Parcel.java:1618)
09-10 18:31:07.180 I/MonoDroid( 6822): 	at android.bluetooth.IBluetoothGatt$Stub$Proxy.readCharacteristic(IBluetoothGatt.java:1022)
09-10 18:31:07.180 I/MonoDroid( 6822): 	at android.bluetooth.BluetoothGatt.readCharacteristic(BluetoothGatt.java:856)
09-10 18:31:07.180 I/MonoDroid( 6822): 	at mono.java.lang.RunnableImplementor.n_run(Native Method)
09-10 18:31:07.180 I/MonoDroid( 6822): 	at mono.java.lang.RunnableImplementor.run(RunnableImplementor.java:30)
09-10 18:31:07.180 I/MonoDroid( 6822): 	at android.os.Handler.handleCallback(Handler.java:761)
09-10 18:31:07.180 I/MonoDroid( 6822): 	at android.os.Handler.dispatchMessage(Handler.java:98)
09-10 18:31:07.180 I/MonoDroid( 6822): 	at android.os.Looper.loop(Looper.java:156)
09-10 18:31:07.180 I/MonoDroid( 6822): 	at android.app.ActivityThread.main(ActivityThread.java:6523)
09-10 18:31:07.180 I/MonoDroid( 6822): 	at java.lang.reflect.Method.invoke(Native Method)
09-10 18:31:07.180 I/MonoDroid( 6822): 	at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:941)
09-10 18:31:07.180 I/MonoDroid( 6822): 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:831)
09-10 18:31:07.180 I/MonoDroid( 6822): 
An unhandled exception occured.

09-10 18:31:12.701 E/mono    ( 6822): 
09-10 18:31:12.701 E/mono    ( 6822): Unhandled Exception:
09-10 18:31:12.701 E/mono    ( 6822): Java.Lang.NullPointerException: Attempt to invoke interface method 'java.util.Iterator java.lang.Iterable.iterator()' on a null object reference
09-10 18:31:12.701 E/mono    ( 6822):   at (wrapper dynamic-method) System.Object.f36023d6-e2a3-4dab-bac1-a422ff3fd481(intptr,intptr)
09-10 18:31:12.701 E/mono    ( 6822):   --- End of managed Java.Lang.NullPointerException stack trace ---
09-10 18:31:12.701 E/mono    ( 6822): java.lang.NullPointerException: Attempt to invoke interface method 'java.util.Iterator java.lang.Iterable.iterator()' on a null object reference
09-10 18:31:12.701 E/mono    ( 6822): 	at android.os.Parcel.readException(Parcel.java:1671)
09-10 18:31:12.701 E/mono    ( 6822): 	at android.os.Parcel.readException(Parcel.java:1618)
09-10 18:31:12.701 E/mono    ( 6822): 	at android.bluetooth.IBluetoothGatt$Stub$Proxy.readCharacteristic(IBluetoothGatt.java:1022)
[... cut ...]
09-10 18:31:12.702 E/mono-rt ( 6822): 	at android.os.Looper.loop(Looper.java:156)
09-10 18:31:12.702 E/mono-rt ( 6822): 	at android.app.ActivityThread.main(ActivityThread.java:6523)
09-10 18:31:12.702 E/mono-rt ( 6822): 	at java.lang.reflect.Method.invoke(Native Method)
09-10 18:31:12.702 E/mono-rt ( 6822): 	at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:941)
09-10 18:31:12.702 E/mono-rt ( 6822): 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:831)
09-10 18:31:12.702 E/mono-rt ( 6822): 
```

aritchie, 2018-09-10:
You'll have to wait until I have time to look into it then

commit 0436da67 (2018-09-11): Fix #232 - marshalling was not trapping error on droid
closed 2018-09-11 by aritchie