aritchie/bluetoothle #233: Attempt to unregister already unregistered input channel. Channel is unrecoverably broken.

hubert-associates, 2018-09-06:
## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [] iOS
 - [x] Android
 - [] tvOS
 - [] macOS
 - [] .NET Standard
 - UWP IS NOT SUPPORTED DURING THE BETA PERIOD - ONLY PRs ARE WELCOME FOR NOW

## Version of OS(s) listed above with issue
4 to 8.1

## Version of Library
Newest: 6.2.0

## Expected Behaviour
After update from library v3.0.2 to 6.2.0 I get the Android errors:

Attempt to unregister already unregistered input channel. Channel is unrecoverably broken.

This happens when a device has been successfully connecte and is transmitting fine. When I turn the Bluetooth device off, I used to get a clean status-change callback with state DISCONNECTED. Now I get the above error and the app crashes (is disposed).

I am using the same devices to test, no changes there (android 7 phones).

Any idea what I need to change or differences between 3.0.2 and 6.2.0 here? 

## Actual Behavior
_If applicable, please include entire exception - type, message, stacktrace_
Above messages from LogCat on Phone, I'll try to supply from VStudio ASAP:

## Code of Conduct
* No Library Update Requests
* No Timeline Requests
* Please read through the documentation and FAQs
* If you have a complaint about the docs, FIX IT - DO NOT FILE IT HERE!  I won't guarantee everything is always 100% up-to-date since I am just 1 person.
* Please be respectful of my time. I put a lot of effort into my libraries.  I try to support the issue requests within a day.  I'm not psychic, so please take the time to fill out the above form as complete as possible

aritchie, 2018-09-06:
> Any idea what I need to change or differences between 3.0.2 and 6.2.0 here?

Please note that this is not a general support channel which questions like the above question is abusing.  You need to send some sort of sample and/or FULL stack trace for me to work with BEFORE filing an issue.  I don't want to be rude, but I provide code, not day-to-day support.

hubert-associates, 2018-09-06:
I know! If I had a full stack trace I would be really happy too. The traces from above (debug mode) is currently the most info I get.  Can you tell me how to put your lib in verbose/debug mode? Or do I have to use your sources instead of the lib. Just checking to see if anybody had had this behavior too... I continue to debug of course. Thanks!

hubert-associates, 2018-09-06:
Here is the stack trace, best I get at the moment:

06 13:46:06.619 D/ClBle1-1( 4895): Pre try uploadToBlobStorage.
09-06 13:46:06.639 D/ClBle1-1( 4895): pre Update History List.
09-06 13:46:11.525 D/BluetoothGatt( 4895): readCharacteristic() - uuid: 96964059-2f52-47b1-844f-0c1002b36ba5
09-06 13:46:11.579 D/BluetoothGatt( 4895): onCharacteristicRead() - Device=2C:5A:8D:0B:1F:41 handle=770 Status=0
09-06 13:46:11.581 D/BluetoothManager( 4895): getConnectionState()
09-06 13:46:11.581 D/BluetoothManager( 4895): getConnectedDevices
09-06 13:46:11.606 D/ClBle1-1( 4895): Pre try uploadToBlobStorage.
09-06 13:46:11.633 D/ClBle1-1( 4895): pre Update History List.
09-06 13:46:16.526 D/BluetoothGatt( 4895): readCharacteristic() - uuid: 96964059-2f52-47b1-844f-0c1002b36ba5
09-06 13:46:16.590 D/BluetoothGatt( 4895): onCharacteristicRead() - Device=2C:5A:8D:0B:1F:41 handle=770 Status=0
09-06 13:46:16.592 D/BluetoothManager( 4895): getConnectionState()
09-06 13:46:16.592 D/BluetoothManager( 4895): getConnectedDevices
09-06 13:46:16.629 D/ClBle1-1( 4895): Pre try uploadToBlobStorage.
09-06 13:46:16.644 D/ClBle1-1( 4895): pre Update History List.
09-06 13:46:21.526 D/BluetoothGatt( 4895): readCharacteristic() - uuid: 96964059-2f52-47b1-844f-0c1002b36ba5
09-06 13:46:21.584 D/BluetoothGatt( 4895): onCharacteristicRead() - Device=2C:5A:8D:0B:1F:41 handle=770 Status=0
09-06 13:46:21.586 D/BluetoothManager( 4895): getConnectionState()
09-06 13:46:21.586 D/BluetoothManager( 4895): getConnectedDevices
09-06 13:46:21.603 D/ClBle1-1( 4895): Pre try uploadToBlobStorage.
09-06 13:46:21.623 D/ClBle1-1( 4895): pre Update History List.
09-06 13:46:26.526 D/BluetoothGatt( 4895): readCharacteristic() - uuid: 96964059-2f52-47b1-844f-0c1002b36ba5
09-06 13:46:26.579 D/BluetoothGatt( 4895): onCharacteristicRead() - Device=2C:5A:8D:0B:1F:41 handle=770 Status=0
09-06 13:46:26.581 D/BluetoothManager( 4895): getConnectionState()
09-06 13:46:26.581 D/BluetoothManager( 4895): getConnectedDevices
09-06 13:46:26.607 D/ClBle1-1( 4895): Pre try uploadToBlobStorage.
09-06 13:46:26.630 D/ClBle1-1( 4895): pre Update History List.
09-06 13:46:31.526 D/BluetoothGatt( 4895): readCharacteristic() - uuid: 96964059-2f52-47b1-844f-0c1002b36ba5
09-06 13:46:31.590 D/BluetoothGatt( 4895): onCharacteristicRead() - Device=2C:5A:8D:0B:1F:41 handle=770 Status=0
09-06 13:46:31.592 D/BluetoothManager( 4895): getConnectionState()
09-06 13:46:31.592 D/BluetoothManager( 4895): getConnectedDevices
09-06 13:46:31.613 D/ClBle1-1( 4895): Pre try uploadToBlobStorage.
09-06 13:46:31.648 D/ClBle1-1( 4895): pre Update History List.
09-06 13:46:36.527 D/BluetoothGatt( 4895): readCharacteristic() - uuid: 96964059-2f52-47b1-844f-0c1002b36ba5
Unhandled Exception:

System.TimeoutException: The operation has timed out.

Thread finished: <Thread Pool> #9
The thread 0x9 has exited with code 0 (0x0).
09-06 13:46:56.306 D/BluetoothGatt( 4895): onCharacteristicRead() - Device=2C:5A:8D:0B:1F:41 handle=770 Status=133
Thread finished: <Thread Pool> #10
The thread 0xa has exited with code 0 (0x0).
Thread finished: <Thread Pool> #4
The thread 0x4 has exited with code 0 (0x0).
Thread finished: <Thread Pool> #17
The thread 0x11 has exited with code 0 (0x0).
Thread finished: <Thread Pool> #16
The thread 0x10 has exited with code 0 (0x0).
Thread finished: <Thread Pool> #11
The thread 0xb has exited with code 0 (0x0).
Thread started: <Thread Pool> #19
Thread started: <Thread Pool> #20
Thread started: <Thread Pool> #21
Thread started: <Thread Pool> #22
Thread started: <Thread Pool> #23
Thread started: <Thread Pool> #24
Thread started: <Thread Pool> #25
Thread started: <Thread Pool> #26
Thread started: <Thread Pool> #27
Thread started: <Thread Pool> #28
Thread started: <Thread Pool> #29
09-06 13:47:18.560 E/mono    ( 4895): 
09-06 13:47:18.560 E/mono    ( 4895): Unhandled Exception:
09-06 13:47:18.560 E/mono    ( 4895): System.TimeoutException: The operation has timed out.
09-06 13:47:18.560 E/mono    ( 4895):   at System.Reactive.PlatformServices.ExceptionServicesImpl.Rethrow (System.Exception exception) [0x00006] in <053e0001366b482a9efec39df52d1c9c>:0 
09-06 13:47:18.560 E/mono    ( 4895):   at System.Reactive.ExceptionHelpers.Throw (System.Exception exception) [0x0000a] in <053e0001366b482a9efec39df52d1c9c>:0 
09-06 13:47:18.560 E/mono    ( 4895):   at System.Reactive.Stubs+<>c.<.cctor>b__2_1 (System.Exception ex) [0x00000] in <053e0001366b482a9efec39df52d1c9c>:0 
09-06 13:47:18.560 E/mono    ( 4895):   at System.Reactive.AnonymousSafeObserver`1[T].OnError (System.Exception error) [0x0000e] in <053e0001366b482a9efec39df52d1c9c>:0 
[... 25 lines cut ...]
09-06 13:47:18.573 E/mono-rt ( 4895):   at System.Reactive.PlatformServices.ExceptionServicesImpl.Rethrow (System.Exception exception) [0x00006] in <053e0001366b482a9efec39df52d1c9c>:0 
09-06 13:47:18.573 E/mono-rt ( 4895):   at System.Reactive.ExceptionHelpers.Throw (System.Exception exception) [0x0000a] in <053e0001366b482a9efec39df52d1c9c>:0 
09-06 13:47:18.573 E/mono-rt ( 4895):   at System.Reactive.Stubs+<>c.<.cctor>b__2_1 (System.Exception ex) [0x00000] in <053e0001366b482a9efec39df52d1c9c>:0 
09-06 13:47:18.573 E/mono-rt ( 4895):   at System.Reactive.AnonymousSafeObserver`1[T].OnError (System.Exception error) [0x0000e] in <053e0001366b482a9efec39df52d1c9c>:0 
[... cut ...]
09-06 13:47:18.573 E/mono-rt ( 4895):   at System.Reactive.Concurrency.ConcurrencyAbstractionLayerImpl+Timer.Tick (System.Object state) [0x00000] in <053e0001366b482a9efec39df52d1c9c>:0 
09-06 13:47:18.573 E/mono-rt ( 4895):   at System.Threading.Timer+Scheduler.TimerCB (System.Object o) [0x00007] in <43dbbdc147f2482093d8409abb04c233>:0 
09-06 13:47:18.573 E/mono-rt ( 4895):   at System.Threading.QueueUserWorkItemCallback.System.Threading.IThreadPoolWorkItem.ExecuteWorkItem () [0x00008] in <43dbbdc147f2482093d8409abb04c233>:0

hubert-associates, 2018-09-06:
Looks like when the internal connection timeout is longer than the observer's, then an exception is thrown. I've tried to catch the exception but appears to be outside of my code scope. 

Crashes:
`character.ReadInterval(interval).TakeWhile(g => g.Characteristic.Service.Device.Status == ConnectionStatus.Connected).Subscribe(async gattresult =>`

Does not crash:
`character.ReadInterval(interval).Timeout(TimeSpan.FromSeconds(intervalSecs + 2)).TakeWhile(g => g.Characteristic.Service.Device.Status == ConnectionStatus.Connected).Subscribe(async gattresult =>`

aritchie, 2018-09-06:
This is why I have timeouts listed as something you should be using on the readme page.  Also, where are you trapping the error because I don’t see it in the code provided

hubert-associates, 2018-09-07:
Many thanks.  Yes, read that and I have put try-catch blocks everywhere I can think, but the exception is not caught. It seem to bubble around all of my generic catch blocks. Where/how to catch it?   -- I know it is not your thing but a pointer appreciated. I will try to find in your sample again. 

Also on README:  "DO NOT manage reconnection yourself". Well, I am guilty here since every time I turn off the scan I don't get a reconnect after a timeout. i.e. when I power-cycle the device and wait a while in-between, I can't get a reconnect unless the scan is still operative. Crap, what am I missing?

aritchie, 2018-09-07:
I still don't see code so I can only guess again what you aren't doing properly.  My guess is you aren't using the subscribe exception check which is why you have things like the connection status check (which is unnecessary if you check full exceptions).

.Subscribe(
   result => {}, 
   ex => {}, 
   () => {... // completion handler }
);  

>  This was working wit lib version 3.0.2 but probably just luck
Than use v3!!!  This is not a valid argument here.  There is no paid support.

>  "DO NOT manage reconnection yourself". Well, I am still guilty here since every time I turn off the scan I don't get a reconnect after a timeout. i.e. when I power-cycle the device and wait a while in-between, I can't get a reconnect unless the scan is still operative.

Once again, I still have no code from you to even begin to guess what may be going wrong.  I don't know your host device at all.  Maybe it cycles its mac address.... who knows!

> Any pointer appreciated

Unfortunately, you have continually gone outside my "code of conduct" so at this point, I'm going to cut off the support train.  You want tips/pointers, I'm available for short term contracts.

closed 2018-09-07 by aritchie

hubert-associates, 2018-09-09:
Thank you. Not meaning to be outside of anybody's code, just trying to understand how your lib works. Thanks again.