aritchie/bluetoothle #235: Issue with GetKnownCharacteristics not working

ianespana, 2018-09-14:
## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [] iOS
 - [*] Android
 - [] tvOS
 - [] macOS
 - [] .NET Standard
 - UWP IS NOT SUPPORTED DURING THE BETA PERIOD - ONLY PRs ARE WELCOME FOR NOW

## Version of OS(s) listed above with issue
7.1.1 (API 25)

## Version of Library
6.2.1

## Expected Behaviour
Phone should connect to my BLE peripheral and start reading data off it, ideally using notifications
```
        async Task SetupBleDevice()
        {
            Debug.WriteLine("D1");
            var _devices = await _adapter.GetConnectedDevices();
            foreach (var item in _devices)
            {
                item.CancelConnection();
            }
            Debug.WriteLine("D2");
            _selectedItem.Device.Connect();
            Debug.WriteLine("D3");
            //_selectedItem.IsConnected = true;
            //_deviceConnected = true;
            //await _selectedItem.Device.RequestMtu(255);
            Debug.WriteLine("D4");
            IGattCharacteristic tempCharacteristics = await _selectedItem.Device.GetKnownCharacteristics(KnowServices[0], TempReadingCharacteristics);
            Debug.WriteLine("D5");
            await tempCharacteristics.EnableNotifications(true);
            Debug.WriteLine("D6");

            var res = await tempCharacteristics.Read();

            tempCharacteristics.RegisterAndNotify().Subscribe(result =>
            {
                if (result.Characteristic.Uuid == TempReadingCharacteristics[0])
                {
                    Debug.WriteLine("Temp 1");
                }
                else if (result.Characteristic.Uuid == TempReadingCharacteristics[1])
                {
                    Debug.WriteLine("Temp 2");
                }
                else if (result.Characteristic.Uuid == TempReadingCharacteristics[2])
                {
                    Debug.WriteLine("Temp 3");
                }
                else if (result.Characteristic.Uuid == TempReadingCharacteristics[3])
                {
                    Debug.WriteLine("Temp 4");
                }
            });
        }
```

## Actual Behavior
The code gets stuck and doesn't pass GetKnownCharacteristics

## Code of Conduct
* No Library Update Requests
* No Timeline Requests
* Please read through the documentation and FAQs
* If you have a complaint about the docs, FIX IT - DO NOT FILE IT HERE!  I won't guarantee everything is always 100% up-to-date since I am just 1 person.
* Please be respectful of my time. I put a lot of effort into my libraries.  I try to support the issue requests within a day.  I'm not psychic, so please take the time to fill out the above form as complete as possible

aritchie, 2018-09-14:
`_selectedItem.Device.Connect();` <= does not mean device is connected.  Connection could happen hours later.  You can use ConnectWait() if you know the device is in range

ianespana, 2018-09-14:
I also tried it out with ConnectWait(), but it's the same story. It never gets to show the D5 or D6 prints to the console, let alone the ones I have for the notifications

gunpuz, 2018-09-14:
Shouldn't you use timeouts on operations?
Like: `.Timeout(...)`

ianespana, 2018-09-15:
> Shouldn't you use timeouts on operations?
> Like: `.Timeout(...)`

Yes, I had that as well as ToTask() in the first place, but I changed it just to debug

aritchie, 2018-09-15:
I wasn't able to reproduce this.  As next steps, I would try getting all available characteristics and services to verify that the guids match.

ianespana, 2018-09-15:
> I wasn't able to reproduce this. As next steps, I would try getting all available characteristics and services to verify that the guids match.

Alright, I'll try that and report back

ianespana, 2018-09-15:
Ok, so, I tried getting all characteristics using the same method I was using before, but it didn't work either. I then poked around the view and added a behavior to bind ItemTapped to a command, and then handle all the connection in the command, and that fixed it.
Pretty weird...

aritchie, 2018-09-15:
I’m pretty sure you’re doing something “off” here, but if you’re happy with that - best of luck

closed 2018-09-15 by aritchie