aritchie/bluetoothle #236: IsConnectable

gunpuz, 2018-09-14:
## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [] iOS
 - [*] Android
 - [] tvOS
 - [] macOS
 - [] .NET Standard
 - UWP IS NOT SUPPORTED DURING THE BETA PERIOD - ONLY PRs ARE WELCOME FOR NOW

## Version of OS(s) listed above with issue

## Version of Library
6.2.0

## Expected Behaviour

## Actual Behavior
_If applicable, please include entire exception - type, message, stacktrace_

IsConnectable is always false
IsConnectable on Android platforms seems to be not even implemented, well at least not that i can see?

[implementation](https://github.com/aritchie/bluetoothle/blob/c019be11f935d883bee100eebd7f4027927765fb/Plugin.BluetoothLE/Platforms/Android/Internals/InternalScanRecord.cs#L73)
[link](https://github.com/aritchie/bluetoothle/blob/c019be11f935d883bee100eebd7f4027927765fb/Plugin.BluetoothLE/Platforms/Android/AdvertisementData.cs#L27)

## Code of Conduct
* No Library Update Requests
* No Timeline Requests
* Please read through the documentation and FAQs
* If you have a complaint about the docs, FIX IT - DO NOT FILE IT HERE!  I won't guarantee everything is always 100% up-to-date since I am just 1 person.
* Please be respectful of my time. I put a lot of effort into my libraries.  I try to support the issue requests within a day.  I'm not psychic, so please take the time to fill out the above form as complete as possible

aritchie, 2018-09-14:
I have no idea what byte it is in the scan result.  This value is usually useless as an FYI.  A device could still have a GATT regardless of this value.  

In most implementations, you know the end device you intend to connect to which also makes this flag further useless.

gunpuz, 2018-09-14:
well, yes but if more than one person wants to connect to same device, that is pain if you actually don't understand the reason why connection is not made. It would be great to see that someone has taken device. on ios this connectable works.

gunpuz, 2018-09-14:
https://developer.android.com/reference/android/bluetooth/le/ScanResult#isConnectable()

aritchie, 2018-09-14:
I can't use ScanResult out of the box if I want to support android 4x for which I have a need at the moment.  So you'll have to wait until I have time to look what byte has this value.  I'll be honest, it isn't a priority right now.  PR's are always welcome.

gunpuz, 2018-09-14:
Ok, when i will have more time (at en of this week maybe), i will try to do pull request, (never done pull req :)).

gunpuz, 2018-09-14:
Acutally it is pretty simple?
From: https://stackoverflow.com/questions/28467590/detecting-whether-a-ble-device-is-connectable-on-android

```
Prior to Android 8.0, unfortunately it is not possible.

A connectable advertisement is determined by the PDU Header byte 0. You can see this in the example structure below:
The problem is on devices prior to Anroid 8.0, the Android scanning APIs give you no access to these headers. You get exactly three fields in the callback from Android 4.x:
```

So my patch would be something like:

```
if(scanResult.isLegacy()){
    IsConnectable = false; // :(
}
else{
    IsConnectable = scanResult.isConnectable();
}
```

aritchie, 2018-09-14:
I can't use scan result right now as I said.  Please feel free to try this yourself and test it.  

I will look at a later date otherwise.

aritchie, 2019-05-22:
This is now implemented in the replacement library Shiny.BluetoothLE

closed 2019-05-22 by aritchie