aritchie/bluetoothle #237: WhenAnyCharacteristicDiscovered observable does raise the onCompleted action on subscription

michaelkollmann, 2018-09-19:
## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [x] iOS
 - [x] Android
 - [ ] tvOS
 - [ ] macOS
 - [ ] .NET Standard
 - UWP IS NOT SUPPORTED DURING THE BETA PERIOD - ONLY PRs ARE WELCOME FOR NOW

## Version of OS(s) listed above with issue

iOS 11.4.1
Android 8.1.0

## Version of Library

6.2.0

## Expected Behaviour

I expect the onComplete action to be called after all services and characteristics have been discovered

## Actual Behavior

The onComplete action never get's called. I now that all characteristics get discovered by looking at my debug output but the onComplete action just doesn't get called.

Here's my code I use for connecting and getting the characteristics:

```
device.Connect();

// get characteristics
var characteristics = new List<IGattCharacteristic>();
var subscription = device..WhenAnyCharacteristicDiscovered().Subscribe(characteristic =>
{
    // onNext
    Log?.Trace("Discovered the characteristic {0}", characteristic.Uuid);
    characteristics.Add(characteristic);
}, () =>
{
    // onCompleted
    Log?.Trace("Discovered all {0} characteristics", characteristics.Count);
});
```

Am I missing something? Is this behavior intended?
Imho the onCompleted should be raised after all characteristics have been discovered. And I know for certain that discovery works because my debug output lists all characteristics my device has in the onNext action

Thanks!

aritchie, 2018-09-19:
"When" type functions/observables were not meant to complete.  They are meant to fire across reconnections automatically without the need to hook into connection state yourself. 

For characteristics, this is particularly important since the characteristic state is lost across (re)connections

aritchie, 2018-09-19:
You'll notice I do things like this for: 

IDevice.GetKnownCharacteristics  & IDevice.WhenKnownCharacteristicsDiscovered

michaelkollmann, 2018-09-19:
Ok, understood so far. 
Thank you very much for your quick response!

So instead of doing the following:

- Connect to device
- Get all characteristics from device
- Read single characteristic
- Disconnect

I should rather
- Connect to device
- use the "ReadCharacteristic" extension to read the value from the device using the service and characteristic uuid. I should not "care" about getting all characteristics first as the library will handle this for me
- Disconnect

Right?

Sorry for my uninformed questions, I'm still pretty new to all the reactive stuff.

aritchie, 2018-09-19:
You should hook the "When" characteristic methods first, then do Connect.  I noticed you are "pinning" the characteristics in a collection - I would avoid using that list because it is invalid on a disconnect.

CrossBleAdapter.Current.ScanForDevice(uuid or name) (if you know the device is in range or has to be in range) 
or
CrossBleAdapter.Current. GetKnownDevice(uuid) if you know the device

- both methods yield a device
After that

IDevice.WhenKnownCharacteristicDiscovered(serviceUuid, characteristicUuidS).Subscribe(ch =>
{
});

I would suggest taking a look at http://www.introtorx.com/ to get a good handle on the reactive stuff before deep diving through this library.  You can also use the samples and unit tests in this project for help.  

Best of luck

michaelkollmann, 2018-09-20:
Allright, I think I got it!

Thank you for your support and thanks for the great library! 
Best thing out there for cross platform bluetooth development!