aritchie/bluetoothle #238: Support for retrieveConnectedPeripherals on iOS

pravincar, 2018-09-20:
## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [x] iOS
 - [] Android
 - [] tvOS
 - [] macOS
 - [] .NET Standard
 - UWP IS NOT SUPPORTED DURING THE BETA PERIOD - ONLY PRs ARE WELCOME FOR NOW

## Version of OS(s) listed above with issue
All

## Version of Library
All

## Expected Behaviour
On iOS, if my phone has connected to a device via another app, a scan then does not show that device in the list. Using retrieveConnectedPeripherals with a known service UUID would make it possible to find and connect to such a device

## Actual Behavior
_If applicable, please include entire exception - type, message, stacktrace_
getConnectedDevices only shows connections made by my app and not devices already connected.

## Code of Conduct
* No Library Update Requests
* No Timeline Requests
* Please read through the documentation and FAQs
* If you have a complaint about the docs, FIX IT - DO NOT FILE IT HERE!  I won't guarantee everything is always 100% up-to-date since I am just 1 person.
* Please be respectful of my time. I put a lot of effort into my libraries.  I try to support the issue requests within a day.  I'm not psychic, so please take the time to fill out the above form as complete as possible

aritchie, 2018-09-20:
That use-case isn't going to go well for you, but will look into this.

pravincar, 2018-09-20:
Thanks for looking into this @aritchie. Your work on this plugin is great! If you would like me to work on this and submit a PR, I can give this a shot.
This change does require consideration of the API structure so your thoughts on how this should be implemented would be important.

closed 2018-10-05 by aritchie