aritchie/bluetoothle #241: RegisterAndNotify call multiple times (depends on on of reconnect)

alberk8, 2018-09-27:
## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [] iOS
 - [*] Android
 - [] tvOS
 - [] macOS
 - [] .NET Standard
 - UWP IS NOT SUPPORTED DURING THE BETA PERIOD - ONLY PRs ARE WELCOME FOR NOW

## Version of OS(s) listed above with issue
Andoroid 5.0 API 21 - Samsung Note 3

## Version of Library
6.2.1

## Expected Behaviour
When reconnect occurs the RegisterAndNotify should be called once for every write.  The first connection works as intended.

## Actual Behavior
_If applicable, please include entire exception - type, message, stacktrace_
The second time the ble device reconnect the same write command will caused RegisterAndNotify to be called two times, and subsequent reconnect will add on the RegisterAndNotify call until the GATT 133 exception crash the app.

## Code of Conduct
* No Library Update Requests
* No Timeline Requests
* Please read through the documentation and FAQs
* If you have a complaint about the docs, FIX IT - DO NOT FILE IT HERE!  I won't guarantee everything is always 100% up-to-date since I am just 1 person.
* Please be respectful of my time. I put a lot of effort into my libraries.  I try to support the issue requests within a day.  I'm not psychic, so please take the time to fill out the above form as complete as possible

```
 int cnt = 1;
 ble = CrossBleAdapter.Current.GetKnownDevice(Guid.Parse("00000000-0000-0000-0000-c156dc001c9b"))
	.Subscribe(gkd =>
	{
		try
		{
			gkd.Connect(new ConnectionConfig() { AndroidConnectionPriority = ConnectionPriority.High, AutoConnect = true });
			gkd.WhenConnected().Subscribe(wch =>
			{

				wch.GetKnownCharacteristics(Guid.Parse("00001523-1212-efde-1523-785123123123"), Guid.Parse("00001524-1212-efde-1523-785123123123"))
				.Subscribe(gkc =>
				{
						gkc.RegisterAndNotify().Subscribe(ran =>
						{
							Console.WriteLine("####### Register and Notify " + Convert.ToBase64String(ran.Data));
						});
	
					cnt += 1;

					gkc.Write(GetDataPart1()).Subscribe(wrr =>
					{
						Console.WriteLine("####### Write Data: " + Convert.ToBase64String(wrr.Data));
					});
					gkc.Write(PowerOffDevice()).Subscribe();
				});
		});

			gkd.WhenDisconnected().Subscribe(wd =>
			{
				Console.WriteLine("##### WhenDisconnected Status: ");
			});
		}
		catch (Exception ex)
		{
			Console.WriteLine("##### Error Out: " + ex.Message);
		}
	});
```

pravincar, 2018-09-27:
I believe you should also be disposing the subscriptions you have created on each connection. A composite disposable and DisposeWith extension is helpful with this.
http://www.jaylee.org/post/2011/04/04/Reactive-Being-fluent-with-CompositeDisposable-and-DisposeWith.aspx

alberk8, 2018-09-27:
> I believe you should also be disposing the subscriptions you have created on each connection. A composite disposable and DisposeWith extension is helpful with this.
> http://www.jaylee.org/post/2011/04/04/Reactive-Being-fluent-with-CompositeDisposable-and-DisposeWith.aspx

I am new to this reactive programming.  Where do I place the DisposeWith? .

pravincar, 2018-09-27:
Okay. Since you have just one notification subscription to dispose you can keep it simple. 
Try something like

````
int cnt = 1;
IDisposable notificationDisposable;
 var ble = CrossBleAdapter.Current.GetKnownDevice(Guid.Parse("00000000-0000-0000-0000-c156dc001c9b"))
	.Subscribe(gkd =>
	{
		try
		{
			gkd.Connect(new ConnectionConfig() { AndroidConnectionPriority = ConnectionPriority.High, AutoConnect = true });
			gkd.WhenConnected().Subscribe(wch =>
			{

				wch.GetKnownCharacteristics(Guid.Parse("00001523-1212-efde-1523-785123123123"), Guid.Parse("00001524-1212-efde-1523-785123123123"))
				.Subscribe(gkc =>
				{
						notificationDisposable = gkc.RegisterAndNotify().Subscribe(ran =>
						{
							Console.WriteLine("####### Register and Notify " + Convert.ToBase64String(ran.Data));
						});
	
					cnt += 1;

					gkc.Write(GetDataPart1()).Subscribe(wrr =>
					{
						Console.WriteLine("####### Write Data: " + Convert.ToBase64String(wrr.Data));
					});
					gkc.Write(PowerOffDevice()).Subscribe();
				});
		});

			gkd.WhenDisconnected().Subscribe(wd =>
			{
				notificationDisposable.Dispose();
				Console.WriteLine("##### WhenDisconnected Status: ");
			});
		}
		catch (Exception ex)
		{
			Console.WriteLine("##### Error Out: " + ex.Message);
		}
	});
````
Disclaimer: I'm don't have much experience with this either

alberk8, 2018-09-27:
Not working either.  I have tried the blog method and it does not seems to dispose.  BTW, the WhenDisconnected be called before any connection is established, essentially disposing the RegisterAndNotify method.

aritchie, 2018-09-27:
@Alberk8 while I appreciate that you are new to reactive programming, please note that this is not a support site at all.  I am going to open up paid consulting in the near future.

There is indeed a problem with your code which I will comment on later today when I have some more time.  You should look through the extension methods from the device as you can literally do what you need in 4 lines of code

aritchie, 2018-09-27:
You essentially are continuously hooking the characteristic notification without ever cleaning it up.  You should either be chaining methods or use some of the extensions found in:

https://github.com/aritchie/bluetoothle/blob/master/Plugin.BluetoothLE/Extensions_Device.cs
https://github.com/aritchie/bluetoothle/blob/master/Plugin.BluetoothLE/Extensions_Characteristics.cs

Also, WhenDisconnected is firing right away because almost all observables in this library fire an initial state and that is by design.  In this case, the device is disconnected so it is returning true right away.  If you don't want that initial state, use Skip(1).  

I am going to strongly encourage you to look into the source code to get an understanding of what's going on under the hood.  Again, this will be the last of any such free support I offer as it is way too much of time suck at this point.  You use OSS at your own discretion.  If you have a legit defect with a full reproducible sample or feature request, I will still be looking into those things as necessary.

closed 2018-09-27 by aritchie