aritchie/bluetoothle #244: System.ArgumentNullException when awaiting Device.ReadCharacteristic

RPHEngineering, 2018-10-01:
## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [x] iOS
 - [?] Android
 - [] tvOS
 - [] macOS
 - [] .NET Standard
 - UWP IS NOT SUPPORTED DURING THE BETA PERIOD - ONLY PRs ARE WELCOME FOR NOW

## Version of OS(s) listed above with issue

11.4

## Version of Library

6.2.2

## Expected Behaviour

There is currently a crash when awaiting Device.ReadCharacteristic which I did not expect ;)

## Actual Behavior

Source: 

```
        protected async Task readMacAddress()
        {
            IGattCharacteristic waka;
            await (Device.ReadCharacteristic(BleService.DeviceInformation, BleCharacteristic.MacAddress).Subscribe(
                onNext: Result =>
                {
                    MacAddress = BitConverter.ToString(Result.Data).Replace("-", string.Empty);
                },
                onError: Error =>
                {
                    throw new BleException("Cannot Retrive MAC Address, Error Reading MAC Address");
                },
                onCompleted: () =>
                {
                    return;

            }) as IObservable<CharacteristicGattResult>);
        }
```

Exception: 

```
System.ArgumentNullException: Value cannot be null.
Parameter name: source
  at System.Reactive.Linq.Observable.GetAwaiter[TSource] (System.IObservable`1[T] source) [0x00003] in <370f6a6bb34048878534065376a195cb>:0
  at Lynkd.Models.Peripherals.RPHPeripheral+<readMacAddress>d__13.MoveNext () [0x00012] in /Users/ericbowman/Documents/Development/Mobile/LYNKD/Lynkd/Models/Peripherals/RPHPeripheral.cs:25
  at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw () [0x0000c] in /Library/Frameworks/Xamarin.iOS.framework/Versions/11.12.0.4/src/Xamarin.iOS/mcs/class/referencesource/mscorlib/system/runtime/exceptionservices/exceptionservicescommon.cs:152
  at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess (System.Threading.Tasks.Task task) [0x00037] in /Library/Frameworks/Xamarin.iOS.framework/Versions/11.12.0.4/src/Xamarin.iOS/mcs/class/referencesource/mscorlib/system/runtime/compilerservices/TaskAwaiter.cs:187
  at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification (System.Threading.Tasks.Task task) [0x00028] in /Library/Frameworks/Xamarin.iOS.framework/Versions/11.12.0.4/src/Xamarin.iOS/mcs/class/referencesource/mscorlib/system/runtime/compilerservices/TaskAwaiter.cs:156
  at System.Runtime.CompilerServices.TaskAwaiter.ValidateEnd (System.Threading.Tasks.Task task) [0x00008] in /Library/Frameworks/Xamarin.iOS.framework/Versions/11.12.0.4/src/Xamarin.iOS/mcs/class/referencesource/mscorlib/system/runtime/compilerservices/TaskAwaiter.cs:128
  at System.Runtime.CompilerServices.TaskAwaiter.GetResult () [0x00000] in /Library/Frameworks/Xamarin.iOS.framework/Versions/11.12.0.4/src/Xamarin.iOS/mcs/class/referencesource/mscorlib/system/runtime/compilerservices/TaskAwaiter.cs:113
  at Lynkd.Models.Peripherals.RPHGunbox+<InitializeAsync>d__1.MoveNext () [0x00020] in /Users/ericbowman/Documents/Development/Mobile/LYNKD/Lynkd/Models/Peripherals/RPHGunbox.cs:12
  at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw () [0x0000c] in /Library/Frameworks/Xamarin.iOS.framework/Versions/11.12.0.4/src/Xamarin.iOS/mcs/class/referencesource/mscorlib/system/runtime/exceptionservices/exceptionservicescommon.cs:152
  at System.Runtime.CompilerServices.AsyncMethodBuilderCore+<>c.<ThrowAsync>b__6_0 (System.Object state) [0x00000] in /Library/Frameworks/Xamarin.iOS.framework/Versions/11.12.0.4/src/Xamarin.iOS/mcs/class/referencesource/mscorlib/system/runtime/compilerservices/AsyncMethodBuilder.cs:1018
  at UIKit.UIKitSynchronizationContext+<>c__DisplayClass1_0.<Post>b__0 () [0x00000] in /Library/Frameworks/Xamarin.iOS.framework/Versions/11.12.0.4/src/Xamarin.iOS/UIKit/UIKitSynchronizationContext.cs:24
  at Foundation.NSAsyncActionDispatcher.Apply () [0x00000] in /Library/Frameworks/Xamarin.iOS.framework/Versions/11.12.0.4/src/Xamarin.iOS/Foundation/NSAction.cs:125
  at at (wrapper managed-to-native) UIKit.UIApplication.UIApplicationMain(int,string[],intptr,intptr)
  at UIKit.UIApplication.Main (System.String[] args, System.IntPtr principal, System.IntPtr delegate) [0x00005] in /Library/Frameworks/Xamarin.iOS.framework/Versions/11.12.0.4/src/Xamarin.iOS/UIKit/UIApplication.cs:79
  at UIKit.UIApplication.Main (System.String[] args, System.String principalClassName, System.String delegateClassName) [0x0002c] in /Library/Frameworks/Xamarin.iOS.framework/Versions/11.12.0.4/src/Xamarin.iOS/UIKit/UIApplication.cs:63
  at Lynkd.iOS.Application.Main (System.String[] args) [0x00001] in /Users/ericbowman/Documents/Development/Mobile/LYNKD/Lynkd.iOS/Main.cs:17
```

aritchie, 2018-10-01:
You are not using RX properly at all!!  You can’t subscribe and await like this.  Just await on device.readcharacteristic directly.

closed 2018-10-01 by aritchie

RPHEngineering, 2018-10-02:
Could you give me a little bit more direction than that? I am brand new to RX and I am learning. A sample snippet, or a link to a blog post?

aritchie, 2018-10-02:
Come on.... seriously I'm not google or a teacher.  There are samples & unit tests in this repository for how to use this library and there are a ton of websites on reactive programming.  

http://reactivex.io/

RPHEngineering, 2018-10-02:
For anyone else who comes across this; I was overcomplicating the situation. As aritchie commented, I was not using RX correctly and you can await the ReadCharacteristic call directly. The correct code is as follows: 

`        protected async Task readMacAddress()
        {
            var macResult = await Device.ReadCharacteristic(BleService.DeviceInformation, BleCharacteristic.MacAddress);
            MacAddress = BitConverter.ToString(macResult.Data).Replace("-", string.Empty);
        }`

Aritchie, you could have been a lot nicer about your help. I'm sure you've been new to something and felt insecure about it before and I hope when you were in that situation others didn't treat you the way you just treated me. That said; I've really enjoyed using your library so thank you for writing and maintaining it.