aritchie/bluetoothle #249: NullReferenceException on AdapterContext.GetDevice

FalmaJeus, 2018-10-04:
Hello.
I have a question about iOS crash on AdapterContext.GetDevice.

Here is the stacktrace.
Application.Main (System.String[] args)
System.NullReferenceException: Object reference not set to an instance of an object

AdapterContext.GetDevice (CoreBluetooth.CBPeripheral peripheral)
Adapter.GetKnownDevice (System.Guid deviceId)
BleDeviceService+<Connect>d__28.MoveNext ()
AsyncMethodBuilderCore+<>c.<ThrowAsync>b__6_0 (System.Object state)
NSAsyncSynchronizationContextDispatcher.Apply ()
(wrapper managed-to-native) UIKit.UIApplication.UIApplicationMain(int,string[],intptr,intptr)
UIApplication.Main (System.String[] args, System.String principalClassName, System.String delegateClassName)
Application.Main (System.String[] args)

I have checked your source code, and i can see following source code on Plugin.BluetoothLE\Platforms\Apple\AdapterContext.cs file.

public IDevice GetDevice(CBPeripheral peripheral) => this.peripherals.GetOrAdd(
            peripheral.Identifier.ToString(),
            x => new Device(this, peripheral)
        );

I wonder that the code is trying to get Identifier of peripheral and if the peripheral is null, then the code would crash. So I think there should be an null handling.
Not sure if this is the reason, but please help me.

Thanks, Falma.

aritchie, 2018-10-04:
Please use the full issue template in the future.

I should put a safety null check before this level, however, this is happening due to something else downstream or you wouldn’t be getting null.  Secondly, you should error trapping on this.

Please provide a full test case that I can evaluate and I can proceed from there

FalmaJeus, 2018-10-04:
Ok, I will use issue template in the future.

Once the app is connected to bluetooth device, the app saves its Guid.
On next launch, it loads the saved Guid and connect to that bluetooth device.
I think if that device already disappeared, the NullReferenceException will come up.

Please let me know if you have more questions.

closed 2018-10-05 by aritchie