aritchie/bluetoothle #252: PauseBetweenInvocations, what does it means?

Magic73, 2018-10-23:
Hi Archie,
on Android devices, I see that you use a queue to process commands.
There's a parameter which is used to delay between operations, but I don't understand why it should be used.

PauseBetweenInvocations: This performs pauses between each operation helping android recover from itself

what do you mean, when you write "recover from itself" ?

I'm using the library to control a slotcar racing system, and I need the maximum responsiveness. Which are the drawbacks on settings this parameter to 0 ms, instead of the default 100ms?

In the older version of the library, this delay wasn't present (as the quueue system too).

Cristiano

aritchie, 2018-10-23:
The queue and this pause variable is all mechanisms to prevent race conditions within some of the android implementations.  You can disable it at your own risk.  If you start getting GATT errors, you’ll need to reenable.

closed 2018-10-23 by aritchie

Magic73, 2018-10-23:
About Android implementations, do you mean OS version? Marshmellow, Nougat, Oreo.. ?

aritchie, 2018-10-24:
Different manufacturers, OS versions, etc.  I would suggest googling about android gatt issues.

I don't know about the exceptions since I have no idea what the exceptions are.  

Please note that this form is for defects with reproducible cases and feature requests.

Magic73, 2018-10-24:
Ok, but where we can discuss about this? (if we can…)

anyway, reading the android code.. in DeviceContext you have a ProcessQueue.

Inside it, you have a loop which dequeue the pending writing task. Inside the loop, the task is run with:

await task();

but then, the writing method perform the writing inside:
this.context.InvokeOnMainThread(() =>
            {
… WRITING
}

so the task ends 99% of the times, before the writing is really done. 
your sleep, should prevent concurrent writing due to this "delay".

but, why do you need to call the writing inside the main thread? and if it's mandatory, in order to prevent concurrent issue.. why don't you run ProcessQueue loop inside the main thread ??

this will solve any issue, and allow to not have any custom delay.

Magic73, 2018-10-24:
You have the latency due to the PauseBetweenInvocations.
Right now, I set PauseBetweenInvocations to 0ms, and I commented the InvokeOnMainThread call.

aritchie, 2018-10-24:
@Magic73 It is true that any crash inside of that handle will kill the app, but all invocations within it are error trapped.  I've done an extreme amount of testing around this.  If you find one, report it and I'll fix it, but do your research of the code before making a statement like that please.  As for WriteWithoutResponse, you can't tell when it is complete anyhow since there is no callback.  Lastly, I don't need to comment on every thing I do in the library.  I'm not paid for this time.  You can study the code and google for yourself why I do these things.  I'm not here to answer repeated questions around the design.  

Guys, seriously - if you don't like the checks - turn them off.  There are flags for all of this, but don't bring forward any bugs if you disable them.  I've done the research on various flavours of android and this setup works in almost all of the cases I've seen.

Lastly, if you don't like that I'm doing this or this library - you are free to either submit PR's or use another library.

aritchie, 2018-10-24:
Cristiano

I'm not insulted.  I've been very clear about what I'm willing to do for this library.  You give me a full stacktrace and/or a reproducible sample, I will fix the problem.  You have stated you have errors yet I have no errors listed here.  I even entertain feature requests if they suit the project.

However, I'm not here to discuss the decisions I've made or why.  I'm happy to answer a few questions, but not the barrage you've sent, nor am I willing to get into a debate about those answers.  I'm also not willing to engage in day-to-day support questions similar to what you've asked in the past because it is my time at that point.  If you need that level of support, we can talk about consulting rates & contracts.

Also, your pull request is unnecessary.  As I've already stated, you can turn it off.  I'll even add a flag to remove the queue if there is enough reason, but I'll limit support with those flags changed from their defaults.

CrossBleAdapter.AndroidConfiguration.ShouldInvokeOnMainThread = false;
CrossBleAdapter.AndroidConfiguration.PauseBetweenInvocations = null;

Magic73, 2018-10-24:
Hi Allan,
here one of the stack trace, where the app crash. as you may see.. it crash in the library, and I have no way to handle it.

Unhandled Exception:

Plugin.BluetoothLE.BleException: Error during characteristic write

10-24 16:36:35.143 D/Mono    (15516): DllImport attempting to load: '/system/lib64/liblog.so'.
10-24 16:36:35.149 D/Mono    (15516): DllImport loaded library '/system/lib64/liblog.so'.
10-24 16:36:35.149 D/Mono    (15516): DllImport searching in: '/system/lib64/liblog.so' ('/system/lib64/liblog.so').
10-24 16:36:35.150 D/Mono    (15516): Searching for '__android_log_print'.
10-24 16:36:35.150 D/Mono    (15516): Probing '__android_log_print'.
10-24 16:36:35.150 D/Mono    (15516): Found as '__android_log_print'.
10-24 16:36:35.164 I/MonoDroid(15516): UNHANDLED EXCEPTION:
10-24 16:36:35.183 I/MonoDroid(15516): Plugin.BluetoothLE.BleException: Error during characteristic write ---> Plugin.BluetoothLE.BleException: Failed to write to characteristic
10-24 16:36:35.183 I/MonoDroid(15516):   at Plugin.BluetoothLE.GattCharacteristic+<>c__DisplayClass6_0.<WriteWithoutResponse>b__0 (System.IObserver`1[T] ob) [0x00095] in C:\Progetti\ScalextricArcPro\BLE\Plugin.BluetoothLE\Platforms\Android\GattCharacteristic.cs:53 
10-24 16:36:35.183 I/MonoDroid(15516):    --- End of inner exception stack trace ---
10-24 16:36:35.183 I/MonoDroid(15516):   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess (System.Threading.Tasks.Task task) [0x0003e] in <43dbbdc147f2482093d8409abb04c233>:0 
10-24 16:36:35.183 I/MonoDroid(15516):   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification (System.Threading.Tasks.Task task) [0x00028] in <43dbbdc147f2482093d8409abb04c233>:0 
10-24 16:36:35.183 I/MonoDroid(15516):   at System.Runtime.CompilerServices.TaskAwaiter.ValidateEnd (System.Threading.Tasks.Task task) [0x00008] in <43dbbdc147f2482093d8409abb04c233>:0 
10-24 16:36:35.183 I/MonoDroid(15516):   at System.Runtime.CompilerServices.ConfiguredTaskAwaitable`1+ConfiguredTaskAwaiter[TResult].GetResult () [0x00000] in <43dbbdc147f2482093d8409abb04c233>:0 
10-24 16:36:35.183 I/MonoDroid(15516):   at Plugin.BluetoothLE.Internals.DeviceContext+<>c__DisplayClass19_1`1+<<Invoke>b__1>d[T].MoveNext () [0x0006a] in C:\Progetti\ScalextricArcPro\BLE\Plugin.BluetoothLE\Platforms\Android\Internals\DeviceContext.cs:74 
10-24 16:36:35.183 I/MonoDroid(15516): --- End of stack trace from previous location where exception was thrown ---
10-24 16:36:35.183 I/MonoDroid(15516):   at System.Reactive.PlatformServices.ExceptionServicesImpl.Rethrow (System.Exception exception) [0x00006] in <370f6a6bb34048878534065376a195cb>:0 
10-24 16:36:35.183 I/MonoDroid(15516):   at System.Reactive.ExceptionHelpers.Throw (System.Exception exception) [0x0000a] in <370f6a6bb34048878534065376a195cb>:0 
10-24 16:36:35.183 I/MonoDroid(15516):   at System.Reactive.Stubs+<>c.<.cctor>b__2_1 (System.Exception ex) [0x00000] in <370f6a6bb34048878534065376a195cb>:0 
10-24 16:36:35.183 I/MonoDroid(15516):   at System.Reactive.AnonymousSafeObserver`1[T].OnError (System.Exception error) [0x0000e] in <370f6a6bb34048878534065376a195cb>:0 
10-24 16:36:35.183 I/MonoDroid(15516):   at System.Reactive.Linq.ObservableImpl.Take`1+Count+_[TSource].OnError (System.Exception error) [0x00000] in <370f6a6bb34048878534065376a195cb>:0 
10-24 16:36:35.183 I/MonoDroid(15516):   at System.Reactive.AutoDetachObserver`1[T].OnErrorCore (System.Exception exception) [0x00000] in <370f6a6bb34048878534065376a195cb>:0 
10-24 16:36:35.183 I/MonoDroid(15516):   at System.Reactive.ObserverBase`1[T].OnError (System.Exception error) [0x0001c] in <370f6a6bb34048878534065376a195cb>:0 
10-24 16:36:35.183 I/MonoDroid(15516):   at Plugin.BluetoothLE.Internals.DeviceContext+<>c__DisplayClass19_1`1+<<Invoke>b__1>d[T].MoveNext () [0x0010d] in C:\Progetti\ScalextricArcPro\BLE\Plugin.BluetoothLE\Platforms\Android\Internals\DeviceContext.cs:81 
10-24 16:36:35.183 I/MonoDroid(15516): --- End of stack trace from previous location where exception was thrown ---
10-24 16:36:35.183 I/MonoDroid(15516):   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess (System.Threading.Tasks.Task task) [0x0003e] in <43dbbdc147f2482093d8409abb04c233>:0 
10-24 16:36:35.184 I/MonoDroid(15516):   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification (System.Threading.Tasks.Task task) [0x00028] in <43dbbdc147f2482093d8409abb04c233>:0 
10-24 16:36:35.184 I/MonoDroid(15516):   at System.Runtime.CompilerServices.TaskAwaiter.ValidateEnd (System.Threading.Tasks.Task task) [0x00008] in <43dbbdc147f2482093d8409abb04c233>:0 
10-24 16:36:35.184 I/MonoDroid(15516):   at System.Runtime.CompilerServices.TaskAwaiter.GetResult () [0x00000] in <43dbbdc147f2482093d8409abb04c233>:0 
10-24 16:36:35.184 I/MonoDroid(15516):   at Plugin.BluetoothLE.Internals.DeviceContext+<ProcessQueue>d__24.MoveNext () [0x00087] in C:\Progetti\ScalextricArcPro\BLE\Plugin.BluetoothLE\Platforms\Android\Internals\DeviceContext.cs:193 
10-24 16:36:35.184 I/MonoDroid(15516): --- End of stack trace from previous location where exception was thrown ---
10-24 16:36:35.184 I/MonoDroid(15516):   at System.Runtime.CompilerServices.AsyncMethodBuilderCore+<>c.<ThrowAsync>b__6_0 (System.Object state) [0x00000] in <43dbbdc147f2482093d8409abb04c233>:0 
[... cut ...]
10-24 16:36:37.159 E/mono-rt (15516):   at System.Runtime.CompilerServices.TaskAwaiter.GetResult () [0x00000] in <43dbbdc147f2482093d8409abb04c233>:0 
10-24 16:36:37.159 E/mono-rt (15516):   at Plugin.BluetoothLE.Internals.DeviceContext+<ProcessQueue>d__24.MoveNext () [0x00087] in C:\Progetti\ScalextricArcPro\BLE\Plugin.BluetoothLE\Platforms\Android\Internals\DeviceContext.cs:193 
10-24 16:36:37.159 E/mono-rt (15516): --- End of stack trace from previous location where exception was thrown ---
10-24 16:36:37.159 E/mono-rt (15516):   at (wrapper dynamic-method) System.Object.34(intptr,intptr)

Magic73, 2018-10-24:
> Cristiano
> I'm not insulted.  I've been very clear about what I'm willing to do for this library.  You give me a full stacktrace and/or a reproducible sample, I will fix the problem.  You have stated [...]

I did more test, and removing invokeonmainthread and pausebetweeninvocation, raise some exception in some of the writewithoutresponse. Instead reading and writing doesn't have any issue (invokeonmainthread is used in reading method too, and I commented it also).

So, I "moved" the Task.Delay from ProcessQueue to WriteWithoutResponse method.

just after: 

` ob.Respond( new CharacteristicGattResult(this, value));`

I place: 

```
var ts = CrossBleAdapter.AndroidConfiguration.PauseBetweenInvocations;
if (ts.TotalMilliseconds > 0)
         Task.Delay(ts).Wait();
```

Now, I'm doing after some stress test, I found that 10ms is enough to avoid that calling

`this.context.Gatt.WriteCharacteristic(this.native)`

return false, raising an exception.

aritchie, 2018-10-24:
Ok so we're all over the place in terms of topics here, so let's start with the error.  I did add error trapping to methods.  Are you catching them in your subscribe?

For the pause, it can be added to the WriteWithoutResponse, however I won't be taking anything that involves dropping any existing functionality, sorry.  I will add the flag back for dropping out the queue feature, but that's about it.

10-15ms isn't really going to be the timeframe used within .NET.  It will most likely be firing on 40-50ms in reality, but at those low values, nothing is guaranteed.  Whatever values you discover should be configured on your project as it isn't a "1 size fits all"

Magic73, 2018-10-24:
WriteWithoutResponse doesn't have an external action which handle the error..

This is an example on how I call the normal write:

```
characteristic.Write(new byte[] { 1 }).Take(1).Subscribe(result =>
{
	System.Diagnostics.Trace.WriteLine($"Command Sent!");
},
(exc) => { 
	System.Diagnostics.Trace.WriteLine($"Command exception!");
});
```

and this is how I call the WriteWithoutResponse:

```
try
{
	characteristic.WriteWithoutResponse(new byte[] { 1 }).Take(1).Subscribe(); <-- it doesn't have the exception action
	System.Diagnostics.Trace.WriteLine($"Command sent!");
}
catch (Exception exc)
{
	System.Diagnostics.Trace.WriteLine($"Command exception!");
}
```

aritchie, 2018-10-24:
All Subscribe calls have an overload for error trapping including WriteWithoutResponse. WriteWithoutResponse still going into the queue so it does still run a callback with an error or completion.

aritchie, 2018-10-24:
There is no callback for when WriteWIthResponse completes.  Time is the only thing if you are going to call rapidly into it.  Use Write if you are going to call in rapidly.  Trap exceptions in both.  This is why all of the configurable parameters exist.  

At this point, I'm not even sure what we're solving anymore and I've answered all of the questions I'm willing to answer for it, so I'm going to lock this issue.  I will release a patch in the future that allows you to disable the queue.  Best of luck