aritchie/bluetoothle #253: Can't connect to device on Android

JasperJian101, 2018-10-24:
## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [] Android

## Version of OS(s) listed above with issue

## Version of Library

## Expected Behaviour
Connect to device

## Actual Behavior
Nothing happens when I use device.Connect() or device.PairingRequest().

I am using an NRF52 as the target device. I am able to scan for it but cannot connect to it.
I've been trying to use this library for the past 3 hours. I am able to scan for devices and the devices show up but when I try to connect to a device, nothing happens. Am I missing something here?

aritchie, 2018-10-24:
I'm missing so much info to even begin to help you.  Please submit reproducible sample. 

Please don't comment on the time you've spent or general questions as "Am I missing something here?".  Those are unnecessary and can be deemed as disrespectful.  Remember, this isn't a paid library - you use it at your own discretion.

JasperJian101, 2018-10-24:
My Android version is 6.0

JasperJian101, 2018-10-24:
Nothing happens when I call those functions to connect to device. THis is the code I use to connect: 
        public void ConnectDevice(IDevice device)
        {
            device.Connect(new GattConnectionConfig { IsPersistent = true, Priority = ConnectionPriority.Normal, });
            device.WhenAnyCharacteristicDiscovered().Subscribe(characteristic => {
                if(characteristic.Uuid == ServerToClientCharacGuid)
                {
                    ServerToClientCharacteristic = characteristic;
                    ServerToClientCharacteristic.SubscribeToNotifications();
                    ServerToClientCharacteristic.WhenNotificationReceived().Subscribe(result => {
                        RaiseServerToClientPacketReceived(result.Data);
                    });
                }
                if(characteristic.Uuid == ClientToServerCharacGuid)
                {
                    ClientToServerCharacteristic = characteristic;
                }
            });
        }

JasperJian101, 2018-10-24:
So I was previously using another Xamarin BLE library however, I found issues with it so I tried to use your one. I know what is meant to happen on the target device when I try to connect as I am debugging on the target device, but nothing happens on the target device when I try to connect to it.

JasperJian101, 2018-10-24:
The library version is the latest

aritchie, 2018-10-24:
There isn't anything wrong with your code that wouldn't cause it to connect that I can see.  Is there any logs in console?

aritchie, 2018-10-24:
Since connections can happen "hours" later, there is also a Device.WhenConnectionFailed() for when it errors or you can use "await device.ConnectWait()" that may help

gunpuz, 2018-10-26:
Some devices like HUAWEI does not like autoconnect, in some specific cases. Give a try with connection config autoconnect = false

aritchie, 2018-10-28:
No response for user

closed 2018-10-28 by aritchie