aritchie/bluetoothle #256: Device is not disconnecting propertly if luck is not on your side

gunpuz, 2018-10-30:
## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [????] iOS
 - [*] Android
 - [] tvOS
 - [] macOS
 - [] .NET Standard
 - UWP IS NOT SUPPORTED DURING THE BETA PERIOD - ONLY PRs ARE WELCOME FOR NOW

## Version of OS(s) listed above with issue

## Version of Library

## Expected Behaviour

## Actual Behavior
_If applicable, please include entire exception - type, message, stacktrace_

So there is problem with device disconnect. Sometimes it is disconnecting, sometimes it doesn't.
More specific:
Problem is with function ConnectWait() over [here](https://github.com/aritchie/bluetoothle/blob/21c1c351bf2e187d30a971f54ef3c75199a3f9bd/Plugin.BluetoothLE/Extensions_Device.cs#L68)
And more specific:
With check that device is not connected: over [here](https://github.com/aritchie/bluetoothle/blob/21c1c351bf2e187d30a971f54ef3c75199a3f9bd/Plugin.BluetoothLE/Extensions_Device.cs#L21)

So, when you do something like this:
```
dev.connect();
await dev.connectWait(); 
```
Device connection is called sometimes 2 times. It depends, if device has managed to connect before connectWait checks

So something i guess goes wrong with resource cleanup in android and device happens to be disconnected.

What i would expect is - some checks if device connection call is already pending in [here](https://github.com/aritchie/bluetoothle/blob/21c1c351bf2e187d30a971f54ef3c75199a3f9bd/Plugin.BluetoothLE/Extensions_Device.cs#L21)
Or at least some info that, you cannot use both commands at the same time

## Code of Conduct
* No Library Update Requests
* No Timeline Requests
* Please read through the documentation and FAQs
* If you have a complaint about the docs, FIX IT - DO NOT FILE IT HERE!  I won't guarantee everything is always 100% up-to-date since I am just 1 person.
* Please be respectful of my time. I put a lot of effort into my libraries.  I try to support the issue requests within a day.  I'm not psychic, so please take the time to fill out the above form as complete as possible

aritchie, 2018-10-30:
This is not a “bad problem” with the library.  Connect and ConnectWait are not meant to both be called.

Connect issues a request that could actually happen in 10 days.

ConnectWait means to fire that request and wait for the connected callback

closed 2018-10-30 by aritchie

gunpuz, 2018-10-30:
But if connectWait is ment to be called seperatly, why you can't give any connection configuration like you would do in connect(config) call?
And would it not be more clear to call ConnectWait  as ConnectAndWait

aritchie, 2018-10-30:
Listen - there are lots of things that could be done differently, but this is not a paid library.

gunpuz, 2018-10-30:
And sure, the description of ConnectWait says it all
```
        /// <summary>
        /// Waits for connection to actually happen
        /// </summary>
        /// <param name="device"></param>
        /// <returns></returns>
        public static IObservable<IDevice> ConnectWait(this IDevice device)
```

aritchie, 2018-10-30:
I'm not sure if that is your attempt at snark, but please note there is ZERO tolerance for it if it is.  

If you don't like something, you can a) live with it, b) contribute via docs, code, etc, c) there will be a paid support option in the very near future, or d) find another library.  

In any case, this is not an issue and while I could do things to protect the user from themselves, that is not the goal of this library.