aritchie/bluetoothle #257: On iOS send a WriteWithoutResponse when Peripheral is ready

pravincar, 2018-10-31:
## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [x] iOS
 - [] Android
 - [] tvOS
 - [] macOS
 - [] .NET Standard
 - UWP IS NOT SUPPORTED DURING THE BETA PERIOD - ONLY PRs ARE WELCOME FOR NOW

## Version of OS(s) listed above with issue
iOS 11+

## Version of Library
6.2.2

## Expected Behaviour
When using WriteWithoutResponse and writing faster than iOS can dispatch, the plugin could wait and send out the request when the Peripheral.IsReadyToSendWriteWithoutResponse is fired. It appears that this was implemented before and then removed when files were split instead of using #ifdef.
https://github.com/aritchie/bluetoothle/blob/49c145db37763b795c6a7a5deb07f3620391cd65/Plugin.BluetoothLE/Platforms/Apple/GattCharacteristic.cs#L47

## Actual Behavior
_If applicable, please include entire exception - type, message, stacktrace_
Right now the plugin, switches to WriteWithResponse when the buffer is full and I'm not able to find a work around to achieve a high data rate.

## Code of Conduct
* No Library Update Requests
* No Timeline Requests
* Please read through the documentation and FAQs
* If you have a complaint about the docs, FIX IT - DO NOT FILE IT HERE!  I won't guarantee everything is always 100% up-to-date since I am just 1 person.
* Please be respectful of my time. I put a lot of effort into my libraries.  I try to support the issue requests within a day.  I'm not psychic, so please take the time to fill out the above form as complete as possible

aritchie, 2018-11-02:
Turns out, there was a reason I pulled this - it didn't work well.

aritchie, 2018-11-09:
The swap off is actually a bug which I'm definitely going to remove.  The original request is with IsReadyToSendWriteWithoutResponse is what I will likely end up passing on.

aritchie, 2018-11-13:
@raindeer Please note that timeline requests are not welcome here.  I'll release it soon.

raindeer, 2018-11-13:
Oh sry! My bad. Thanks for a great lib :)

commit 79405209 (2018-11-18): fix #257 - writewitoutresponse shouldn't be swapping to write when unavailable.  Stub comment IsReadyToWrite implementation for later date
closed 2018-11-18 by aritchie