aritchie/bluetoothle #258: sample does not compile on latest Xamarin Studio 7.16.10

ormaa, 2018-11-02:
## Version of OS(s) listed above with issue

MacOS 10.14
Version of project taken on the 2nd of November 2018.

Sample application cannot be compiled, and cannot be run on android device, or IOS device.

Xamarin Studio report an error during compilation : 

/Users/olivier/.nuget/packages/microsoft.build.tasks.git/1.0.0-beta-63127-02/build/Microsoft.Build.Tasks.Git.targets(5,5): Error MSB4018: The "Microsoft.Build.Tasks.Git.LocateRepository" task failed unexpectedly.
System.TypeInitializationException: The type initializer for 'Microsoft.Build.Tasks.Git.TaskImplementation' threw an exception. ---> System.IO.FileNotFoundException: Could not load file or assembly 'Microsoft.Build.Tasks.Git.Operations' or one of its dependencies
  at System.AppDomain.Load (System.Reflection.AssemblyName assemblyRef, System.Security.Policy.Evidence assemblySecurity) [0x001a9] in /Users/builder/jenkins/workspace/build-package-osx-mono/2018-02/external/bockbuild/builds/mono-x64/mcs/class/corlib/System/AppDomain.cs:777 
  at (wrapper remoting-invoke-with-check) System.AppDomain.Load(System.Reflection.AssemblyName,System.Security.Policy.Evidence)
  at System.AppDomain.Load (System.Reflection.AssemblyName assemblyRef) [0x00000] in /Users/builder/jenkins/workspace/build-package-osx-mono/2018-02/external/bockbuild/builds/mono-x64/mcs/class/corlib/System/AppDomain.cs:714 
  at (wrapper remoting-invoke-with-check) System.AppDomain.Load(System.Reflection.AssemblyName)
  at System.Reflection.Assembly.Load (System.Reflection.AssemblyName assemblyRef) [0x00000] in /Users/builder/jenkins/workspace/build-package-osx-mono/2018-02/external/bockbuild/builds/mono-x64/mcs/class/corlib/System.Reflection/Assembly.cs:600 
  at Microsoft.Build.Tasks.Git.TaskImplementation..cctor () [0x0006b] in /_/src/Microsoft.Build.Tasks.Git/TaskImplementation.cs:32 
   --- End of inner exception stack trace ---
  at Microsoft.Build.Tasks.Git.LocateRepository.Execute () [0x0000f] in /_/src/Microsoft.Build.Tasks.Git/LocateRepository.cs:27 
  at Microsoft.Build.BackEnd.TaskExecutionHost.Microsoft.Build.BackEnd.ITaskExecutionHost.Execute () [0x00023] in /_/src/Build/BackEnd/TaskExecutionHost/TaskExecutionHost.cs:631 
  at Microsoft.Build.BackEnd.TaskBuilder+<ExecuteInstantiatedTask>d__26.MoveNext () [0x00212] in /_/src/Build/BackEnd/Components/RequestBuilder/TaskBuilder.cs:787  (MSB4018) (Plugin.BluetoothLE)

aritchie, 2018-11-02:
Please note, that I don't usually support issues with the samples. However, it looks like you're new to the ecosystem.

1) Xamarin Studio is quite old and won't support many things.  You will need to use far newer versions now dubbed Visual Studio for Mac or Rider from JetBrains.  

2) That being said, this project uses something called multitargeting (the library itself, not the sample) which doesn't compile on VS4mac since they haven't added that feature set.  

3) The error you're seeing is a secondary library I use called sourcelink which allows you to debug this library with remote source.  This also doesn't compile in xamarin studio or VS4mac.  Rider also has an issue with it though I have fixed that in the dev branch.

So - moral of the story - use VS for windows or JetBrains Rider /w the dev branch

ormaa, 2018-11-03:
Hi,

Thanks for the answer.

I am working on MAC, with MACOS.
I use Visual studio latest version (sorry, I call it always Xamarin…)

Anyway, if to does not work on MAC, no problem, I will search for something else.
Just, please, add this to your project GitHub, it could help someone else who could be in my case :o)

Congratulation for your work, it seem to be amazing.

thanks
Olivier

aritchie, 2018-11-03:
You can use the actual library in your code if you install from nuget.  It will work fine on macOS.  

Compiling the samples and library yourself can only be compiled using jetbrains rider which is available on Mac as well or VS on windows.

I will add something to the FAQ

closed 2018-11-03 by aritchie