aritchie/bluetoothle #259: Device scan

gunpuz, 2018-11-03:
## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [] iOS
 - [*] Android
 - [] tvOS
 - [] macOS
 - [] .NET Standard
 - UWP IS NOT SUPPORTED DURING THE BETA PERIOD - ONLY PRs ARE WELCOME FOR NOW

## Version of OS(s) listed above with issue

## Version of Library

6.2.2
## Expected Behaviour

First scan works like any other scan calls

## Actual Behavior
_If applicable, please include entire exception - type, message, stacktrace_

It seems that toggling scan on and off for many times can cause it to not to scan again. It is just example, in real world the "rescan" can have bigger time gap between scans for this to happen.
What i am getting is that "Debug.WriteLine("### Last scan is working fine");" is never printed.
What could i be doing wrong with the scanning? Is this code any correct?
```
            var tcs = new TaskCompletionSource<bool>();
            var scan = Plugin.BluetoothLE.CrossBleAdapter.Current.Scan().Subscribe(scanResult =>
            {
                // try if scan is even working
                Debug.Write(scanResult);
                tcs.TrySetResult(true);
            });
            await tcs.Task;
            tcs = null;
            scan.Dispose();
            Debug.WriteLine("### First scan is working fine");

            for (var i = 0; i < 5; i++)
            {
                // just toggle scan on and off
                var scanner = Plugin.BluetoothLE.CrossBleAdapter.Current.Scan().Subscribe(scanResult =>
                {
                    
                });
            
                scanner.Dispose(); // to stop scanning
            }

            Debug.WriteLine("### Check if the last scan is working");
            tcs = new TaskCompletionSource<bool>();
            Plugin.BluetoothLE.CrossBleAdapter.Current.Scan().Subscribe(scanResult =>
            {
                Debug.Write(scanResult);
                tcs.TrySetResult(true);
            });
            await tcs.Task;

            Debug.WriteLine("### Last scan is working fine");
```

## Code of Conduct
* No Library Update Requests
* No Timeline Requests
* Please read through the documentation and FAQs
* If you have a complaint about the docs, FIX IT - DO NOT FILE IT HERE!  I won't guarantee everything is always 100% up-to-date since I am just 1 person.
* Please be respectful of my time. I put a lot of effort into my libraries.  I try to support the issue requests within a day.  I'm not psychic, so please take the time to fill out the above form as complete as possible

gunpuz, 2018-11-03:
looks like you should not call Dispose, as in [docs](https://github.com/aritchie/bluetoothle/blob/master/docs/adapter.md). But instead you should call CrossBleAdapter.Current.StopScan(); instead

aritchie, 2018-11-03:
You should call Dispose always.  StopScan is an emergency brake.

On android, you have to give the scanner a second to breathe or the internals can “breakdown” to the point where you have to power cycle the adapter, but you almost have to be trying to break it

The code you have is definitely abusing the adapter and doesn’t appear to be a valid test case.

It sounds like you just want “rescan”?  I’m just not understanding your use case here?

gunpuz, 2018-11-03:
StopScan() seems to work FINE at least on android.
Yes, well, in my code example i try to break it, but it there is problems in real case scenarios too. 
The same symptoms after few scans it is not returning any devices.

StopScan and Dispose don't want to work together

About "rescan" - yes i want to scan, then stop, do some connections to devices, read something, and scan again.

aritchie, 2018-11-03:
Dispose doesn’t always call the cleanup immediately.  The cleanup is identical otherwise.  It is the way RX works.

If you are happy with StopScan, go ahead and use it.  If you are claiming “real world problems”, please submit a more realistic case that I can test against.  If you really are hammering the adapter, there is only so much I can and am willing to do.

I do something similar in an app I have.  I don’t scan quite as much as you appear to, but it works.

gunpuz, 2018-11-03:
Well, yes for me, in the "real world scenarios" it depends - sometimes it is just fine and sometimes it hangs - but the biggest problem is that it seems to be blocking next scans. If one scan is "corruped", then with high probability next scans will be too.

The one "real" scenario:
You have 2 pages and both of them use Scan.
When you open first page, you start scanning. Before navigating away to Second page, you call dispose
But the second the Second page opens you start to scan. The time between Dispose and new scan can be quite small. And it sometimes hung up like i said, and doesn't return any devices.
The other scenarios - it just hungs up at random

gunpuz, 2018-11-03:
I added pull request, because Dispose it is not working 100% for me. And if it is not reliable, it should not be used?

aritchie, 2018-11-03:
It isn’t blocking scans.  You essentially have crashed the android library itself.  You have to cycle the power on the adapter to free it.  It isn’t this library hampering you.

As for reproducible case, it needs to be a code sample please.  

There are things I can do to fix problems with android, but you will very likely have to change up your case here a bit as there may not be anything the library can do about it.

aritchie, 2018-11-03:
Dispose should be used.  I’ve explained this already.  Dispose can take a second to run.  If StopScan is working for you, great use it, but you have a somewhat unique use case

aritchie, 2018-11-03:
I’m going to close this as StopScan does fix things for you.  Again, I can’t fix all use cases within the library nor am I willing to invest the time to try.

closed 2018-11-03 by aritchie