aritchie/bluetoothle #264: Can't use this package from Nuget

thecashit, 2018-11-13:
Hello there!

I tried to add this lib as nuget package to Xamarin.Forms project.

After adding i have following error message:
<project name>/iOS/MTOUCH: Error MT2001: Could not link assemblies. Reason: Error while processing references of '<project name>, Version=0.0.0.0, Culture=neutral, PublicKeyToken=null' (MT2001) (LOOK_in.iOS)

I tried to reinstall package, change version, and edit references - error still prevent app to be compilled.

Could you please suggest what to do to make it work?

Thnx.

Visual Studio for Mac 7.6.11

aritchie, 2018-11-13:
No idea.  Never seen the problem before.  The error doesn’t even say what the reference missing is.  Please supply a small reproducible sample and I’ll take a look

Also, please use the issue template in the future.

thecashit, 2018-11-13:
Sorry for template issue.

I tried to start new blank project to illustrate problem, but something wrong with it.
I placed only this lines in line according readme: 
```
var server = CrossBleAdapter.Current.CreateGattServer();
var service = server.AddService(Guid.NewGuid(), true);

```
And get:
`Error CS1061: 'IObservable<IGattServer>' does not contain a definition for 'AddService' and no extension method 'AddService' accepting a first argument of type 'IObservable<IGattServer>' could be found (are you missing a using directive or an assembly reference?) (CS1061) (TestApp.iOS)`

Maybe I'm trying to combine a boa and a rhinoceros.
In the description it is written that the REACTIVE plug-in - maybe I have the wrong "type of project"

aritchie, 2018-11-13:
The docs are slightly out of date.  CreateServer is an observable, not a simple return.  

I’m going to strongly caution you, if you haven’t used reactive extensions before, you will have a tough time.

thecashit, 2018-11-13:
Thank you, I will try to sort out the question and come back in a few days. There is a possibility that the problem is in my use of the functional. Yes, I looked in the source that the use has changed and the description does not correspond to reality.

The only reason why I had to get into REACTIVE was that I could not find more nuget packages with support for the BLE server besides yours.
Several clients but no servers with advertising functionality.

aritchie, 2018-11-13:
You need to either await or subscribe to CreateGattServer().  The downstream effect is the compiler being stupid.  Your core project isn't compiling because you have the syntax wrong.  

Best of luck with your project.

closed 2018-11-13 by aritchie

billyjacobs2014, 2020-01-13:
I am trying to use the library but I am getting an error on await server.Start();
Code is as follows:
var server = await CrossBleAdapter.Current.CreateGattServer();
......
await server.Start(new AdvertisementData
{
    LocalName = "TestServer"
});

Error: 
Severity	Code	Description	Project	File	Line	Suppression State
Error	CS1929	'IGattServer' does not contain a definition for 'Start' and the best extension method overload 'Qbservable.Start(IQbservableProvider, Expression<Action>)' requires a receiver of type 'IQbservableProvider'	_4TDS.Mobile.Portable	C:\Development\ara\4TDS\ccs-mobile\Portable\Components\Dashboard\DashboardViewModel.cs	346	Active