aritchie/bluetoothle #265: How to wait for a connection properly

nilslahr, 2018-11-14:
## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [] .NET Standard

## Version of OS(s) listed above with issue
Windows 10

## Version of Library
6.2.2

## Expected Behaviour
I am not sure what is expected.  When calling device.Connect() the device goes into a state of Connecting, but never comes out.  If my device wakes up in the first 4-5 seconds, then the state changes to Connected.  However if it wakes up after 5+ seconds, the state is still Connecting forever and no matter what happens with the real device, it never will connect.

I tried calling "device.Connect()" every 5 seconds or so, but this causes a different issue.  After calling it 2-3 times in a row, the device.NativeDevice = null and at that point the device is no longer valid and I can't do anything with it.

I read all of the docs and reviewed the code. 

a) What would you suggest to wait for the device to connect, looping so that whenever it wakes up the computer connects to the device?

b) Are there any situations when the device.NativeDevice == null after having already used the device.  When I call device.CancelConnection() for example, then NativeDevice = null and at that time I have no way to call device.Connect or do anything as it appears the device object is no longer valid.  

My use case:
1. for every known device, try to connect, continue trying
2. when the device comes alive we connect to it

[image]

aritchie, 2018-11-14:
You don’t (and shouldn’t) call connect repeatedly.  It is a “request to connect”.  It could happen now, it could happen an hour from now.  If you know the device is in range, you can use ConnectWait with a timeout.  Also, you really shouldn’t be touching the Native object unless you have a very specific use case.

If this is for UWP, please read the issue template as I don’t support it.  Secondly, please note that this is not a general support forum.  If you need something beyond the docs, samples, and source - I am available for consultancy.

nilslahr, 2018-11-14:
In fact I am calling/testing ConnectWait with the same results.  When using just Connect, it didn't seem to work like that on iOS for us.  If the device didn't connect within < 10 seconds, then it wouldn't connect after longer periods of time and the Status seems to be "Connecting" forever.  We are using the library mostly on iOS/Android, but we are testing on UWP as well.  Do we need to avoid UWP due to known issues?

aritchie, 2018-11-16:
I'll need to see some code in order to figure out what may be going on.  Too many ways things can be done to even make a guess at what may be going on. 

As for UWP is a platform I simply don't support due to a shortcomings in the UWP API.  It mostly works, I just don't want the hassle personally for a product space that I don't see a ton of traction on.

Thanks for the coffees.  Very much appreciated.

closed 2018-11-26 by aritchie