aritchie/bluetoothle #266: WhenStatusChanged returns incorrect status after adapter is turned off/on

pravincarvalho, 2018-11-20:
## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [x] iOS
 - [] Android
 - [] tvOS
 - [] macOS
 - [] .NET Standard
 - UWP IS NOT SUPPORTED DURING THE BETA PERIOD - ONLY PRs ARE WELCOME FOR NOW

## Version of OS(s) listed above with issue
iOS 12.1

## Version of Library
6.2.2

## Expected Behaviour
Steps to reproduce
1. Connect to BLE device
2. When the device is connected, turn off phone bluetooth adapter from the OS settings
3. Turn off the BLE device (or put it in a state where connections are not possible)
4. Turn on the phone bluetooth adapter.
WhenStatusChanged should be fired with Status == Connecting

## Actual Behavior
_If applicable, please include entire exception - type, message, stacktrace_
WhenStatusChanged is fired with Status == Connected.
The convenience method device.WhenConnected() also returns immediately due to this issue.

## Code of Conduct
* No Library Update Requests
* No Timeline Requests
* Please read through the documentation and FAQs
* If you have a complaint about the docs, FIX IT - DO NOT FILE IT HERE!  I won't guarantee everything is always 100% up-to-date since I am just 1 person.
* Please be respectful of my time. I put a lot of effort into my libraries.  I try to support the issue requests within a day.  I'm not psychic, so please take the time to fill out the above form as complete as possible

aritchie, 2018-11-20:
Make sure background mode for peripherals is enabled.  Also note, you may need to hook to whenstaterestored

aritchie, 2018-11-20:
Please submit code as well.  Explanations are really not sufficient with this library

aritchie, 2018-11-20:
Just for clarification, is the device actually connected at the end of this - you are just expecting to see a Connecting event?

pravincarvalho, 2018-11-20:
The device is not connected at the end of this and I'm expecting to see a Connecting event.
I have hooked into WhenStateRestored and also enabled the BLE background mode.

I will make a simple reproducible code sample and share it with you.

aritchie, 2018-11-20:
You should be listening for the connected event only.  Connecting is a bonus that isn’t always guaranteed especially in background scenarios.

Android & iOS are 2 completely different animals so try not to compare

pravincarvalho, 2018-11-20:
Agreed. The event that I'm currently seeing is Connected when my device is off. Its the end of the day here so I'll investigate further and send you a code sample tomorrow.
Thanks for your quick response.

aritchie, 2018-11-23:
Ok the problem here is fundamentals.  That method is going to be trimmed if your app goes to the background.  Only the delegate will remains.  You need to subscribe to the delegate and store the subscription externally.  I don’t see WhenStateRestored being hooked anywhere.  Also, for issues like this, you really need to submit a small project to see all of the moving parts.

In any case, this is outside a scope of an issue based on what I’m seeing.  If you submit a better end to end taste case, I’m happy to take a look, but right now this looks more of a support case which I don’t do.  If you need something beyond this, I do offer consulting.

pravincarvalho, 2018-11-24:
From looking into the code in more detail, I believe this behaviour is because of 
https://github.com/aritchie/bluetoothle/blob/b46ad0db9d4b79af2a1427b41281d4f6003e78ed/Plugin.BluetoothLE/Platforms/Apple/Device.cs#L141-L142
I'm not sure I understand what the purpose of these lines are, but this would lead WhenStatusChanged to emit the previous state on a new subscription.

aritchie, 2018-11-26:
I don't want to go into potential "what ifs" on this.  I did give you some feedback on the initial issue with your approach.  If you can give me a reproducible sample with the feedback in place, I am happy to take a look or even fix an issue where applicable

closed 2018-11-29 by aritchie