aritchie/bluetoothle #267: [Android]Device Name Returning Null

FabriBertani, 2018-11-26:
## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [] iOS
 - [x] Android
 - [] tvOS
 - [] macOS
 - [x] .NET Standard
 - UWP IS NOT SUPPORTED DURING THE BETA PERIOD - ONLY PRs ARE WELCOME FOR NOW

## Version of OS(s) listed above with issue
Android Lollipop and above
## Version of Library
6.2.3

## Expected Behaviour
Get device name on scan

## Actual Behavior
Actually only get the UUID number and the ManufacturerData.

## Code of Conduct
* No Library Update Requests
* No Timeline Requests
* Please read through the documentation and FAQs
* If you have a complaint about the docs, FIX IT - DO NOT FILE IT HERE!  I won't guarantee everything is always 100% up-to-date since I am just 1 person.
* Please be respectful of my time. I put a lot of effort into my libraries.  I try to support the issue requests within a day.  I'm not psychic, so please take the time to fill out the above form as complete as possible

aritchie, 2018-11-26:
In order to get device name, you have to request location permissions.  Take a look at the setup instructions on where to go to do this.

FabriBertani, 2018-11-26:
Hi @aritchie, actually i make a permission request before scanning and location permission are granted

aritchie, 2018-11-26:
Please provide a full reproducible sample with the issue.

aritchie, 2018-11-26:
I just ran it.  It works fine for me.

aritchie, 2018-11-26:
What kind of device are you running on?

FabriBertani, 2018-11-27:
Hi @aritchie i've tested on:
    **Samsung S5** with Android 6.0
    **Nokia TA-1063** with Android 8.1
    **Lenovo PB1-770M** with Android 5.1.1

Wasn't able to get devices names on scan.

closed 2019-03-14 by aritchie