aritchie/bluetoothle #270: No Status Changed Event when Bluetooth is turned off

bluekuen, 2018-11-30:
## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [x] iOS
 - [] Android
 - [] tvOS
 - [] macOS
 - [] .NET Standard
 - UWP IS NOT SUPPORTED DURING THE BETA PERIOD - ONLY PRs ARE WELCOME FOR NOW

## Version of OS(s) listed above with issue
12.0.1

## Version of Library
current stable

## Expected Behaviour
Disconnect Status event is thrown 

## Actual Behavior
Nothing Happens. 
I'm connected to a bluetooth device and when i turn bluetooth off through the slide out quick menu no status changed event happens.
I added all Bluetooth and accesory background modes to the info plist, didn't fix it.
On Android everything works fine.

## Code of Conduct
* No Library Update Requests
* No Timeline Requests
* Please read through the documentation and FAQs
* If you have a complaint about the docs, FIX IT - DO NOT FILE IT HERE!  I won't guarantee everything is always 100% up-to-date since I am just 1 person.
* Please be respectful of my time. I put a lot of effort into my libraries.  I try to support the issue requests within a day.  I'm not psychic, so please take the time to fill out the above form as complete as possible

aritchie, 2018-11-30:
Please provide a small reproducible sample.

aritchie, 2018-12-03:
No reproducible sample or reply.

closed 2018-12-03 by aritchie

bluekuen, 2018-12-03:
Sorry, for late reply. I can't provide you with a reproducible sample, as it's from a production program.
When i'm connected i call:

```
 var nativeDevice = await BluetoothLEManager.Instance.GetDevice(_deviceService.GetDevice().ID);
                _statusChangedSub = bleDevice.NativeDevice.WhenStatusChanged().Subscribe(async connectionStatus =>
                {

                    if (connectionStatus == ConnectionStatus.Disconnected)
                    {
                        logger.Debug("Device Connections Status changed to {connectionsStatus}", connectionStatus);
                        //if top page is TabbedDevicePage => go back async
                        await DisconnectAsync();

                        await NavigationService.NavigateToAsync<MyDevicesViewModel>(true);

                        await UserDialogs.Instance.AlertAsync(AppResources.ConnectionLost);
                    }
                    else
                    if (connectionStatus == ConnectionStatus.Connected)
                    {
                        logger.Debug("Received noew ConnectionStatus: " + connectionStatus);
                    }
                });
```

Unfortunately, the method doesn't get called on iOS when bluetooth is turned off whilst connected.

aritchie, 2018-12-03:
If you can't supply a full test case, then please don't post anything as it could be anything.  It is important to see things like where this is being hooked.  Is it being GC'd... if so why!?   Please remember, this is not for general support.

bluekuen, 2018-12-03:
I don't think it's GC'd, if i go out of range the disconnected event is raised, only when i turn Bluetooth off in settings it isn't raised. Other status changed events also get raised like connected.