aritchie/bluetoothle #276: Issues with GetConnectedDevices

zlittell, 2018-12-17:
I noticed that when a device is connected to, it disappears from the devices the scanner returns.  I am assuming this is intended behavior.  I am trying to use the CrossBleAdapter.Current.GetConnectedDevices()
function to get all the matching connected devices.  I used your code snippet in the docs and it gives me an error.

```
        public async Task<bool> LoadConnectedDevices()
        {
            var devices = CrossBleAdapter.Current.GetConnectedDevices(new Guid(ApplicationServices.CONSTANTS.MAINUUID));

            foreach (var device in devices)
            {
            }

            return await Task.FromResult(true);
        }
```

`foreach statement cannot operate on variables of type 'IObservable<IEnumerable<IDevice>>' because 'IObservable<IEnumerable<IDevice>>' does not contain a public instance definition for 'GetEnumerator'`

What am I doing wrong here?  I apologize if this is something simple that I am just missing.

-Zack-

aritchie, 2018-12-17:
Hi Zack

I'm not really sure what to tell you beyond what the compiler is saying.  The method returns an async observable, not something you can iterate immediately.  There are samples & documentation located within the repository.  I would also suggest reading a little bit about reactive extensions (RX) before getting to deep into this library as things will get a lot more difficult after this.

Please note that I don't offer support here beyond an occasional quick question, but I do offer direct paid consultancy if needed.  Let me know.

zlittell, 2018-12-17:
Yeah that is my bad.  I did some research and realized I just needed to do what I had already been doing.  Thank you for setting me down the path of RX.  

This code worked for me.  I will strip the extra fluff out and update the docs in a pull request
```
            IObservable<IEnumerable<IDevice>> devices = CrossBleAdapter.Current.GetConnectedDevices(new Guid(ApplicationServices.CONSTANTS.MAINUUID));

            devices.Subscribe(deviceResult =>
            {
                //Debug.WriteLine(deviceResult.ToString());
                foreach (var device in deviceResult)
                {
                    Debug.WriteLine(device.Name.ToString());
                }
            });
```

aritchie, 2018-12-18:
Perfect - thanks for the update.  Glad you were able to fix your issue.  

If you do need more help, please feel free to reach out using linkedin or twitter listed here: https://github.com/aritchie/home

closed 2018-12-18 by aritchie