aritchie/bluetoothle #280: response gets truncate

NZ2011, 2018-12-26:
## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [] iOS
 - [] Android 
 - [] .NET Standard
## Version of OS(s) listed above with issue
Working on Visula studio 2017 professional , Windows 10 and Andoid 8.1 API.

## Version of Library
Plugin.BluetoothLE
Installed version : 6.2.0

## Expected Behaviour
There should be same behaviout for both Android and IOS. only one response would be there for one request instead of 2.

## Actual Behavior
_If applicable, please include entire exception - type, message, stacktrace_
When I am executing request  with writeWIthResponse, Getting 2 responses in android for same request. Same I do with IOS , Getting single response for same request.

## Code of Conduct
* No Library Update Requests
* No Timeline Requests
* Please read through the documentation and FAQs
* If you have a complaint about the docs, FIX IT - DO NOT FILE IT HERE!  I won't guarantee everything is always 100% up-to-date since I am just 1 person.
* Please be respectful of my time. I put a lot of effort into my libraries.  I try to support the issue requests within a day.  I'm not psychic, so please take the time to fill out the above form as complete as possible

aritchie, 2018-12-26:
Please provide a reproducible test case.  You've given me absolutely nothing to work with.  

The description of the issue and the title also aren't matching.  Are you having two different issues?  
1) Title Question - what's being truncated???
2) Description Question - Do you mean Write?  What is replying twice - I don't understand your issue.

aritchie, 2018-12-26:
Also please note - this is not a support forum.  If you need help beyond the samples and documentation, I do offer consultancy services.

NZ2011, 2018-12-27:
Hi ,

I am sending command  like below and using write method with notification on. (ToggleNotify method)
var bytes = Encoding.UTF8.GetBytes(command);
 var result = await CharacteristicTX.Write(bytes);

After executing it, I m getting response twice in Android.
for example I am expecting result data like $got,1,1.00000,2.00000\r\n
so in first response I receive **$got,1,1.0000**
and in second response I get  **,2.00000\r\n**

but in case of ios I get whole data in one response .
I am expecting whole string in one response for android as well.
hope this info would be sufficient.
Thanks.

aritchie, 2018-12-27:
The library doesn’t echo commands.  It will only send what you give it, so your issue is not with the library.  I’m assuming by response, you mean the data you are seeing on your peripheral?  

My guess is your MTU is large enough on iOS that your command goes on one write.  IOS auto negotiates the largest MTU possible.  On android, you must do this yourself as the default MTU is 20 bytes.

aritchie, 2018-12-27:
Also, you aren’t providing nearly enough for me to figure out what is going on.  You will need to provide a test case of what I have said above is not your issue.  

That being said, please be respectful of my time, provide as much detail as possible in the future, and know that support requests are not welcome here.  Only properly framed issues or feature requests

aritchie, 2018-12-31:
No response - closing due to insufficient details

closed 2018-12-31 by aritchie